nodemcu/nodemcu-firmware #3419: ADS1115 module does not support more than one channel

eHc-HA, 2021-04-07:
From my point of view is a bug more than a new feature. ADS 115 has 4 channels, but you can only read one.

### Expected behavior
Assuming that  "device:setting" function can configure the four channels independently and keep the settings for each channel, which is not really documented and perhaps not supported, the functions "ads1115.device:read()" and "ads1115.device:startread()" should have a parameter to specify the channel that is to be read, or return the reads for the four channels.

### Actual behavior
Only the last configured channel can be read. 

### Test code

### NodeMCU startup banner
All releases of ADS1115 module

### Hardware

KT819GM, 2021-04-07:
You need to add delay in your code after switching MUX (channel) before reading again. 20 ms is enough.

eHc-HA, 2021-04-07:
Actually, I've solved the problem using the callback function of the first read to change the settings of the channel and launch the second read. But my thought was that you should perform the settings for the channels you want to read, and read them when required, without the need of restting the channel every time you want to read them.
In any case, the way of working for a multi channel configuration should be better documented.

jmd13391, 2021-04-07:
I have had great success placing the ADS1115 in continuous read mode and feeding the "Conversion_Ready" alert to a GPIO interrupt trigger. Control of the ADS1115 is via enabling/disabling the GPIO Interrupt Trigger. No additional timers/delays are required with this approach. Near real-time reading (and switching) of all four channels and even the differential channel pairs is possible using this approach.

eHc-HA, 2021-04-07:
But, if you have more than one channel operating, how do you select the channel to be read?

closed 2022-04-30 by stale[bot]