nodemcu/nodemcu-firmware #3423: attempt to call field 'newBuffer' (a nil value)

lw1995, 2021-04-11:
`ws2812.init()
local i, buffer = 0, ws2812.newBuffer(300, 4); buffer:fill(0, 0, 0, 0); tmr.create():alarm(50, 1, function()
  i = i + 1
  buffer:fade(2)
  buffer:set(i % buffer:size() + 1, 0, 0, 0, 255)
  ws2812.write(buffer)
end)`
I copied the example code on the document directly.
But something went wrong.
Error report：`attempt to call field 'newBuffer' (a nil value)`

nwf, 2021-04-14:
That's curious and certainly seems like something is amiss.  However, it's very hard to guess what as you appear to have ignored our issue template and, in particular, not said which version of NodeMCU is in use nor what modules your firmware has been compiled against.

lw1995, 2021-04-14:
Thank you. I've solved this problem

HHHartmann, 2021-04-14:
The solution is to use the new pixbuf module.

closed 2021-04-14 by HHHartmann

nwf, 2021-04-14:
If selecting `ws2812` and not `pixbuf` generates a firmware image for which `newBuffer` is `nil`, that's definitely a problem.

HHHartmann, 2021-04-15:
Not really a problem. The buffer was just moved out of ws2812. pixbuf has a newBuffer method which can be used now.
The ws2812 write can still be used without pixbuf available. Just use the string interface.

I just think we could have done better on change management.

nwf, 2021-04-15:
Ah, no, I see.  Absent `pixbuf`, `newBuffer` isn't in the rotable, and so is of course `nil`.  I had missed that and thought we were somehow resolving `pixbuf_new_lua` as `NULL` at link time, sorry.

I guess the right thing to have done would have been to have a stub `newBuffer` method that threw an error or something.  Hm,

lw1995, 2021-04-15:
I found that the reason for this error was the missing Pixel Buffer (PixBuf) module. I have solved the problem the day before yesterday.

lw1995, 2021-04-15:
Thank you. I've solved the problem.

HHHartmann, 2021-04-15:
We had a grace period where ws2812.newBuffer was still available but called the pixbuf implementation.
Don't remember if we printed the depreciation message.

HHHartmann, 2021-04-15:
In general we should think about a way to mitigate api changes. Have the old documentation somewhere and/or include migration pathes. And not only for 1 release but forever. There is so much code out there which is for older versions and migrating it is sometimes quite hard.