nodemcu/nodemcu-firmware #3427: wifi.eventmon bug

vCloudSail, 2021-04-25:
function cb1() end
function cb2() end

wifi.sta.connect(cb1)
wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, cb2)

### Expected behavior

The functions cb1 and cb2 are both called

### Actual behavior
Only function cb1 is called
If a callback function is passed after calling connect, eventmon will not trigger the STA_CONNECTED event, so please explain  in the documentation！！！！
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
wifiInit = {}

function staion_GotIp()
    -- dataTransfer.start()
    local ip = wifi.sta.getip()
    print("station got ip:" .. ip)
end
function staion_Connected(result)

    print("ssid: " .. result.SSID .. "\nbssid: " .. result.BSSID .. "\nchannel: " .. result.channel)
    apinfo = wifi.sta.getapinfo()
    print("apinfo: " .. sjson.encode(apinfo))
end

function wifiInit.sta()
    print("start wifi station mode")
    wifi.setmode(wifi.STATION)
    local cfg = {
        ssid = "LittleMin",
        pwd = "19980708.lcm",
        auto = true,
        save = true
    }
    wifi.sta.config(cfg)
    wifi.sta.connect(
        function()
            print("why???")
        end
    )
end

function wifiInit.ap()
end

wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, staion_Connected)
wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, staion_GotIp)

print("[wifiInit] finish")

return wifiInit

```
### NodeMCU startup banner
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: release
	commit: 136e09739b835d6dcdf04034141d70ab755468c6
	release: 3.0.0-release_20210201
	release DTS: 202102010145
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: enduser_setup,file,gpio,i2c,mdns,net,node,sjson,spi,tmr,u8g2,uart,wifi
 build 2021-04-24 03:15 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
ESP8266

KT819GM, 2021-04-25:
In the documentation regarding `wifi.eventmon` there is written: 

"To ensure all WiFi events are caught, the Wifi event monitor callbacks should be registered as early as possible in init.lua. Any events that occur before callbacks are registered will be discarded!"

It's highly possible that you are trying to trigger the STA_CONNECTED event when your MCU is already connected.

vCloudSail, 2021-04-25:
> In the documentation regarding `wifi.eventmon` there is written:
>
> "To ensure all WiFi events are caught, the Wifi event monitor callbacks should be registered as early as possible in init.lua. Any [...]

I've already tried this, and if you don't believe it try doing it in init.lua
[image]

closed 2022-06-12 by stale[bot]