nodemcu/nodemcu-firmware #3431: how to use the rtctime.c code in a sketch with the Arduino IDE

arduz, 2021-05-10:
Hello, i have tried to use the code in https://github.com/nodemcu/nodemcu-firmware/blob/release/app/modules/rtctime.c    in my sketch, but i couldn't find the nesessary libraries ( #include "module.h", #include "lauxlib.h", #include "rtc/rtctime_internal.h", #include "rtc/rtctime.h"). 
My question: Is it even possible to use such code direct in a sketch? Where can I find the named libraries? I want to use the RTC clock functionality (but not only the RTC RAM or timers) on my ESP12S.
Thanks for any answer.

vsky279, 2021-05-10:
Hi, it's not clear what you are trying to achieve. Why would you want to do that?

Anyway I don't think it's a good idea to do this. For example the `lauxlib.h` library is library that provides routines to talk to Lua. You would need to rewrite the code completely to get rid off this but than you will have nothing left.

marcelstoer, 2021-05-11:
One for #1010 

I think you picked the wrong project. This firmware here is for writing Lua apps on ESP8266/ESP32.

closed 2021-05-11 by marcelstoer