nodemcu/nodemcu-firmware #3434: snprintf.c:41:35: fatal error: c99-snprintf/snprintf.c: No such file or directory

vidalouiswang, 2021-05-22:
### Expected behavior
Build successsfully.
### Actual behavior
CC app/platform/sdcard.c
CC app/platform/ucg_nodemcu_hal.c
AR app/platform/
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/libplatform.a
make[2]: Leaving directory '/home/vidawang/nodemcu-firmware/app/platform'
make[2]: Entering directory '/home/vidawang/nodemcu-firmware/app/libc'
DEPEND: CC app/libc/math.c
DEPEND: CC app/libc/snprintf.c
snprintf.c:41:35: fatal error: c99-snprintf/snprintf.c: No such file or directory
 #include "c99-snprintf/snprintf.c"
                                   ^
compilation terminated.
DEPEND: CC app/libc/stdlib.c
DEPEND: CC app/libc/snprintf.c
snprintf.c:41:35: fatal error: c99-snprintf/snprintf.c: No such file or directory
 #include "c99-snprintf/snprintf.c"
                                   ^
compilation terminated.
CC app/libc/stdlib.c
CC app/libc/snprintf.c
snprintf.c:41:35: fatal error: c99-snprintf/snprintf.c: No such file or directory
 #include "c99-snprintf/snprintf.c"
                                   ^
compilation terminated.
../../Makefile:422: recipe for target '.output/eagle/debug/obj/snprintf.o' failed
make[2]: *** [.output/eagle/debug/obj/snprintf.o] Error 1
make[2]: Leaving directory '/home/vidawang/nodemcu-firmware/app/libc'
../Makefile:380: recipe for target '.subdirs' failed
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory '/home/vidawang/nodemcu-firmware/app'
Makefile:380: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
make: Leaving directory '/home/vidawang/nodemcu-firmware'
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU startup banner
You MUST include the firmware startup banner to describe the version you are using. We reserve the right to immediately close any bug that doesn't.

### Hardware
Describe which ESP8266/ESP32 device you use and document any special hardware setup
required to reproduce the problem.

vidalouiswang, 2021-05-22:
And then I add include to the file nodemcu-firmware/app/libc/snprintf.c like this:
[image]
It doesn't work.

pjsg, 2021-05-22:
did you do a gut submodule update?

vidalouiswang, 2021-05-22:
> did you do a gut submodule update?
> […](#)

Thank you, I used "git clone --recurse-submodules -b release https://github.com/nodemcu/nodemcu-firmwa
re.git" and it worked. Thank you.

closed 2021-05-22 by vidalouiswang