nodemcu/nodemcu-firmware #3438: MQTT Bug

chathurangawijetunge, 2021-06-03:
this is related to all branches
the MQTT module works fine as describe in the documentation but if in case the internet get disconnected without loosing wifi connection the 'offline' will not fire. mqtt:close() will not work mqtt:publish() will return true while eating the RAM (heap)
mqtt :connect() will return error as 'already connected'....

nwf, 2021-06-03:
As per #3068, our MQTT module contains a number of known and suspected deficiencies, and this certainly sounds like it's along the same lines.  At this rate, you should probably not imagine that fixes are forthcoming unless you author them.

chathurangawijetunge, 2021-06-04:
I am not a professional but I think mqtt publish time out is not working correctly.
If no pub-ack with time out we can trigger offline event.
Any ideas... any body

chathurangawijetunge, 2021-06-05:
as i sad I am not a professional but.... in MQTT module 

else if(mud-&gt;connState == MQTT_CONNECT_SENT) { // wait for CONACK time out.    NODE_DBG("MQTT_CONNECT timeout.\n");    mud-&gt;connState = MQTT_INIT;    mqtt_socket_do_disconnect(mud);
 in void mqtt_socket_timer(void *arg) line 796 never fires.

so i add 

  if(!node || espconn_status != ESPCONN_OK){
    lua_pushboolean(L, 0);
  } else {
    os_timer_disarm(&mud->mqttTimer);
    os_timer_arm(&mud->mqttTimer, MQTT_SEND_TIMEOUT * 1000, 0);
    lua_pushboolean(L, 1);  // enqueued succeed.
  }
 to static int mqtt_socket_publish( lua_State* L )

and

  if(mud != NULL){
  NODE_DBG("MQTT_CONNECT timeout.\n")
  mqtt_socket_do_disconnect(mud);  
  }  

to void mqtt_socket_timer(void *arg)

now this triggers a MQTT disconnect if no internet connection after 5 seconds

closed 2021-06-08 by chathurangawijetunge