nodemcu/nodemcu-firmware #3439: Memory leak about in the socket

vidalouiswang, 2021-06-04:
### Expected behavior
Memory won't leak.
### Actual behavior
Memory leak after client requested. 
### Test code

```Lua
-- add code here
-- There are four files:

-- tools.lua:

tools = {};

tools.sha1 = function(content)
    return encoder.toHex(crypto.hash("sha1", content));
end
tools.getTime = function()
    local date = rtctime.epoch2cal(rtctime.get());
    return date.year, date.mon, date.day,
           (tostring(date.year) .. tostring(date.mon) .. tostring(date.day));
end

tools.createID = function()
    local n, e, c = node, encoder, crypto;
    local tmp = n.chipid() + n.flashid();
    id = {hex = e.toHex(c.hmac("sha1", tmp, tmp))};
    print("my id is:", id.hex);
end

tools.configWifi = function(tb)
    tb = tb or {};
    wifi.setmode(wifi.STATIONAP);
    local apCfg = {};
    apCfg.ssid = tb.apSsid or "test";
    apCfg.pwd = tb.apPwd or "88888888";
    apCfg.auth = wifi.WPA_WPA2_PSK;
    apCfg.save = false;

    wifi.ap.config(apCfg);

    wifi.ap.setip({
        ip = "192.168.6.1",
        netmask = "255.255.255.0",
        gateway = "192.168.6.1"
    });

    apCfg = nil;

    if tb.staSsid ~= nil then
        local staCfg = {};
        staCfg.ssid = tb.staSsid;
        staCfg.pwd = tb.staPwd;
        staCfg.save = false;
        wifi.sta.config(staCfg);
    end

end

-- myhttp.lua:

myhttp = {};

function myhttp.createServer(timeout)
    local n = net;
    local server = {server = n.createServer(n.TCP, (timeout or 28000))};
    timeout = nil;

    function server:new()
        local this = {};
[... 282 lines cut ...]
            post("/settime/" + date+ "@" + "8888", prefix+"",function(e){
                alert(e);
            })
        }
    </script>
</body>

</html>

```
### NodeMCU startup banner
NodeMCU 3.0.0.0 
	branch: release
	commit: d4ae3c364bd8ae3ded8b77d35745b7f07879f5f9
	release: 3.0.0-release_20210201 +1
	release DTS: 202105102018
	SSL: false
	build type: float
	LFS: 0x40000 bytes total capacity
	modules: adc,coap,crypto,encoder,file,gpio,http,i2c,mdns,net,node,pwm,rtcfifo,rtcmem,rtctime,sjson,sntp,spi,tmr,uart,websocket,wifi
 build 2021-06-01 00:56 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)

### Hardware
I'm using a typical NodeMCU 4MB dev board.
I'm sure resources have been released after used, but every time a client refresh page(to refetch the file index.html) memory will be reduced by 1KB.

vsky279, 2021-06-04:
Hi, this seems to be a bit complex for someone to get into it.
Can you please provide a **Minimal**, Complete, and Verifiable example?

vidalouiswang, 2021-06-05:
> Hi, this seems to be a bit complex for someone to get into it.
> Can you please provide a **Minimal**, Complete, and Verifiable example?

Haha, I found an error in the process of streamlining the code. After the correction, the memory is no longer leaking.
I had edited the code and thank you for your reply.

closed 2021-06-05 by vidalouiswang

vsky279, 2021-06-05:
@vidalouiswang Good to hear that 👍. I think it is usually the case that the leak is likely in Lua code. This is why a minimal example is needed.