nodemcu/nodemcu-firmware #3443: A new HttpServer

vidalouiswang, 2021-06-09:
### Missing feature
The "httpserver.lua" provide a http server to use, but according to my test, it couldn't handle so many situation, like press F5 very fast in browser, it will cause NodeMCU panic and reboot.

### Justification
So I try to write a new http server module, so far its function is good except memory leak.
I not good at lua, I don't know where the problem is. So I put the code write here, see if someone can fix it.

And this is degined to be include into the LFS image.

```Lua
-- HttpServer.lua
return function()
    local Response = {};
    function Response:new(socket, req)
        local this = {};
        setmetatable(this, self);
        self.__index = self;

        this.socket = socket;

        this.isHeaderWrote = nil -- to mark if header has been sent to client(browser)
        this.headers = {} -- headers to write
        this.queue = {} -- to store data to send
        this.events = {} -- to store events and its callback
        this.finished = false -- to mark if need to close after write data to socket
        this.req = req;

        -- a short name of self.socket:send
        this.send = function(data) this.socket:send(data); end

        -- bind sent callback for send next item in the queue 
        this.socket:on("sent", function()
            this:emit("drain"); -- emit the event to outer
            if #this.queue > 0 then
                this.send(table.remove(this.queue, 1)); -- send the item in the fifo
            else
                this:dispose(); -- release resources
            end
        end)
        return this;
    end

    function Response:on(event, fn) -- store the events and its callback
        table.insert(self.events, {e = event, fn = fn});
    end

    function Response:emit(event, data) -- find the event and execute its callback
        if #self.events <= 0 then return; end
        for _, value in pairs(self.events) do
            if value.e == event then
                value.fn(data);
                break
            end
        end
    end

    function Response:setStatusCode(code) -- set the status code for http
        -- this function show be called before wirte
        self.statusCode = code;
    end

    function Response:write(data) -- write data to socket
        if not self.isHeaderWrote then
            -- write the http headers before body 
            self.isHeaderWrote = true;

            -- add all headers to the send queue 
            for _, value in ipairs(self.headers) do
                table.insert(self.queue,
                             (value.name .. ": " .. value.value .. "\r\n"));
[... 198 lines cut ...]
                    };
                    -- save request
                    obj.req = req;

                    -- execute the callback
                    fn(req, Response:new(sck, req));
                end);

                conn:on("disconnection", function(socket)
                    if socket then
                        local obj = findSocket(socket);
                        if obj then
                            obj.res:emit("close");
                        end
                    end
                end)
            end);
        end

        return srv;
    end

    return newHttpServer;
end
```

vidalouiswang, 2021-06-09:
Use:

```Lua
local HttpServer = node.LFS.newHttpServer()();
local server = HttpServer.createServer(3)
server.listen(80, function(req, res)
    if req.url == "/" then req.url = "/index.html"; end
    -- route
    if req.url:find("^/([^/]+)$") then
        local _, _, name = req.url:find("/(.+)");
        if file.exists(name) then
            res:pushFile(name);
        else
            res:finish();
            gc();
        end
    else 
        --...
    end
    local data = "";
    res:on("data", function(d) data = data .. d; end)
    res:on("close", function() print(data); end)
end)
```

vidalouiswang, 2021-06-09:
Actually I would like to build the new http server module in C at first, but forgive me I checked source of NodeMCU, I didn't find when a connection comes in, which function called the "data arrived" function. If someone could give me a sign of that, I will try to build this in C to make my effort to this project.

TerryE, 2021-09-26:
Not an open issue.

closed 2021-09-26 by TerryE