nodemcu/nodemcu-firmware #3446: I²C speed broken on ESP8266 @ ESP-12F @ LOLIN D1 mini

ldericher, 2021-06-20:
Hi!

It seems, NodeMCU on  only supports one I²C speed. I mostly copied this code from the documentation:

```lua
node.setcpufreq(node.CPU160MHZ) -- to support FASTPLUS speed

-- initialize i2c, set pin 1 as sda, set pin 2 as scl
i2cid=0
sda=1
scl=2
i2c.setup(i2cid, sda, scl, i2c.FASTPLUS)

-- user defined function: read 1 byte of data from device
function read_reg(id, dev_addr, reg_addr, length)
    i2c.start(id)
    i2c.address(id, dev_addr, i2c.TRANSMITTER)
    i2c.write(id, reg_addr)
    i2c.start(id)
    i2c.address(id, dev_addr, i2c.RECEIVER)
    c = i2c.read(id, length)
    i2c.stop(id)
    return c
end

s = read_reg(i2cid, 0x50, 0x00, 100);

for i=1,string.len(s) do 
    print(string.byte(s,i)) 
end;
```

With my scope, I can see the whole transaction takes about 25 ms. It doesn't even matter which speed is put in the `i2c.setup` call, as long as it's any positive integer. `1` or `1000000` instead of `i2c.FASTPLUS` results in the same exact speed visible on my oscilloscope.

I created an [equivalent Arduino sketch](https://github.com/nodemcu/nodemcu-firmware/files/6682255/sketch_jun20a-001.zip), which took just over 2 ms to execute. Same hardware, same pins.

I tried this on the cloud-built "release" firmware image of NodeMCU, as well as a local build on Linux of both the "release" and the "dev" branches – same results.

~ LDer

galjonsfigur, 2021-06-20:
If you can build the firmware yourself, you can comment out [this line](https://github.com/nodemcu/nodemcu-firmware/blob/release/app/include/user_config.h#L208) and you should be able to use the new I2C driver that allows FAST and FASTPLUS speeds.

ldericher, 2021-06-20:
> use the new I2C driver

Fair enough.

Tried it, SLOW speed kind of works (but with frequent errors), anything faster does not produce any recognizable waveform enough. Arduino Version works fine without any errors.

No thanks, NodeMCU is not for me … or at least it doesn't cut it for this project.

closed 2021-09-26 by TerryE