nodemcu/nodemcu-firmware #3453: rtctime.dsleep fails with integer argument

lsborg, 2021-08-01:
### Expected behavior
rtctime.dsleep(2808000000) should put microcontroller to sleep

### Actual behavior
Lua error:      stdin:1: bad argument #1 to 'dsleep' (number has no integer representation)
stack traceback:
        [C]: in function 'rtctime.dsleep'
        stdin:1: in main chunk
        [C]: in ?
        [C]: in ?

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
rtctime.dsleep(2808000000)
```
### NodeMCU startup banner
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
        branch: release
        commit: d4ae3c364bd8ae3ded8b77d35745b7f07879f5f9
        release: 3.0.0-release_20210201 +1
        release DTS: 202105102018
        SSL: false
        build type: float
        LFS: 0x10000 bytes total capacity
        modules: adc,encoder,file,gpio,http,net,node,rtctime,sjson,sntp,tmr,uart,wifi
 build 2021-08-01 02:08 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)

### Hardware
Nodemcu ESP8266

#### Other
node.dsleep(2808000000) works fine.

marcelstoer, 2021-08-01:
`rtctime.dsleep()` uses integer internally: https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/rtctime.c#L188. Hence, the maximum is 4294967295 as documented: https://nodemcu.readthedocs.io/en/latest/modules/rtctime/#rtctimedsleep.
`node.dsleep()` on the other hand uses long (64bit) internally: https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/node.c#L171. See the note about the maximum in the docs: https://nodemcu.readthedocs.io/en/latest/modules/node/#nodedsleep

However, as your 2808000000 value is below the 4294967295 maximum I don't understand what's going on here.

lsborg, 2021-08-01:
Thanks for your answer @marcelstoer 

New info, it works with lua 5.1 

### NodeMCU startup banner
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
        branch: release
        commit: d4ae3c364bd8ae3ded8b77d35745b7f07879f5f9
        release: 3.0.0-release_20210201 +1
        release DTS: 202105102018
        SSL: false
        build type: float
        LFS: 0x10000 bytes total capacity
        modules: adc,encoder,file,gpio,http,net,node,rtctime,sjson,sntp,tmr,uart,wifi
 build 2021-08-01 23:28 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

marcelstoer, 2021-08-02:
@jmattsson is there a good reason `rtctime.dsleep` should _not_ be modeled after what we have in `node.dsleep` (using `uint64 us;`)?

https://github.com/nodemcu/nodemcu-firmware/blob/949875d590e771603a73ed3a90928d48ea3dd38b/app/modules/node.c#L182-L189

jmattsson, 2021-08-02:
IIRC (and it's been years since I worked in this area), the sleep time limitation came from the 32bit registers used to configure the chip for wakeup.

marcelstoer, 2021-08-02:
Oh, true, it even says so in the [rtcmem docs](https://nodemcu.readthedocs.io/en/release/modules/rtcmem/). What could be the reason 2808000000 works with Lua 5.1 but not with Lua 5.3?

jmattsson, 2021-08-02:
Based on that error message, it looks to me like either `lua_tounsigned()` is still trying to fit it into a signed 32bit integer (max value 2147483648), or the `lua_isinteger()` is the one doing it. Is there a `lua_isunsigned()` in Lua 5.3?