nodemcu/nodemcu-firmware #3454: Migrate RTD/MkDocs build to Python 3

marcelstoer, 2021-08-04:
I wanted to enable the RTD docs build for @jmattsson's  `dev-esp32-idf4` branch so we can announce that branch in the docs early. The build failed with "Could not find a version that satisfies the requirement ziglang>=0.8.0". Checking [the RTD logs](https://readthedocs.org/api/v2/build/14397085.txt) made me realize we're still using Python 2.

Upgrading to Python 3 (in the RTD settings) worked but [the build failed again](https://readthedocs.org/api/v2/build/14397146.txt) but for a different reason. Maybe it's because the MkDocs version we (or RTD?) use is ancient as well.

jmattsson, 2021-08-05:
Maybe have a look at https://github.com/readthedocs/readthedocs.org/issues/5474? It looks like the same issue we're hitting, and there seems to be solution(s) there :)

marcelstoer, 2021-08-08:
WIP https://nodemcu.readthedocs.io/en/update-rtd-mkdocs/

A couple of [changes](https://github.com/nodemcu/nodemcu-firmware/commit/9f633ef78321fb1ed110eb820203e51a057f860e) were required.

marcelstoer, 2021-08-19:
- [x]  Implemented in the `dev-esp32-idf4` branch, which will land on `dev-esp32` eventually, with #3457.
- [x]  Implemented in the `dev` branch, which will land on `release` eventually, with #3458.

=> done

PR #3457 (2021-08-16): Update RTD MkDocs
Fixes #3454 for the IDF4 branch.

This will have to be carried over to the `dev` branch afterwards (to eventually end up on `release` as well). IMO we should skip doing this for `dev-esp32` as it'll eventually be replaced by the IDF4 branch.

@jmattsson if https://nodemcu.readthedocs.io/en/update-rtd-mkdocs/ looks fine to you then this should be good enough to land.
PR #3458 (2021-08-19): Update RTD MkDocs
Equivalent of #3457 for the `dev` branch. Fixes #3454.
closed 2021-08-19 by marcelstoer