nodemcu/nodemcu-firmware #3465: Relay High On Boot

shakirhussain78652, 2021-09-08:
I am trying to use NODEMCU ESP8266 V2 and a 5v, 4 channel relay module to be turned on wirelessly via a mobile app.
However every time I power on the circuit. All the relays get activated. 
I have tried every pin from D1 TO D7. Still facing the same problem. 
Please help.

chathurangawijetunge, 2021-09-12:
can't tell anything without seeing the code.

vsky279, 2021-09-18:
Isn't it due to pin states during boot?
https://www.esp8266.com/viewtopic.php?f=32&t=16141
> Hey,
> GPIOs 4 and 5 are the only ones that are always high impedance. All others do have internal pull-ups or are even driven low/high during boot.
> 
> GPIOs 3, 12, 13 and 14 pulled HIGH during boot. Their actual state does not influence the boot process.
> 
> GPIOs 0, 1, 2 and 15 are pulled HIGH during boot and also driven LOW for short periods.
> The device will not boot if 0, 1 or 2 is driven LOW during start-up.
> 
> GPIO 16 is driven HIGH during boot, don't short to GND.

marcelstoer, 2021-09-19:
Nothing the NodeMCU firmware can influence.

closed 2021-09-19 by marcelstoer

TerryE, 2021-09-27:
@shakirhussain78652 I drive 4 power SSRs directly from my ESP board.  These require 5V @ 50 mA IIRC.  The way I do this is to use a MSP23008 with its VCC at 5V direct from my 5V PSU.  It interfaces to the ESP via I2C This is at a 3.3V float high reference with both ends doing pull down to GND as necessary.  The MSP23008 works fine with 0/3.3V signalling.  An no startup pulse issues.