nodemcu/nodemcu-firmware #3482: bug in send_email_smtp.lua

chathurangawijetunge, 2021-12-24:
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: dev
	commit: b91368594db34932dc3d54f74f9fde1e2d6096c1
	release: 
	release DTS: 202112061238
	SSL: true
	build type: float
	LFS: 0x40000 bytes total capacity
	modules: bit,encoder,file,gpio,http,mqtt,net,node,rtctime,sjson,sntp,tmr,uart,wifi,tls
 build 2021-12-19 06:34 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
------------------------------------------
220 smtp.gmail.com ESMTP c13sm7421052pfv.20 - gsmtp
250 smtp.gmail.com at your service
530 5.7.0 Must issue a STARTTLS command first. c13sm7421052pfv.20 - gsmtp
502 5.5.1 Unrecognized command. c13sm7421052pfv.20 - gsmtp
502 5.5.1 Unrecognized command. c13sm7421052pfv.20 - gsmtp
530 5.7.0 Must issue a STARTTLS command first. c13sm7421052pfv.20 - gsmtp
530 5.7.0 Must issue a STARTTLS command first. c13sm7421052pfv.20 - gsmtp
530 5.7.0 Must issue a STARTTLS command first. c13sm7421052pfv.20 - gsmtp
502 5.5.1 Unrecognized command. c13sm7421052pfv.20 - gsmtp
502 5.5.1 Unrecognized command. c13sm7421052pfv.20 - gsmtp
502 5.5.1 Unrecognized command. c13sm7421052pfv.20 - gsmtp
554 5.7.0 Too Many Unauthenticated commands. c13sm7421052pfv.20 - gsmtp

nwf, 2021-12-27:
This is not a bug report, this is a nodemcu version and half of a conversation with a SMTP server.

chathurangawijetunge, 2021-12-28:
yes, not a bug but example code error....
how to send STARTTLS command first.

chathurangawijetunge, 2021-12-28:
yes it is not a firmware bug, but it is a example code error....

HHHartmann, 2021-12-28:
Seems like the example does not support tls but the server requires it.
Maybe just state the limitation to the documentation of the example. 
Please use our template next time.

closed 2021-12-29 by chathurangawijetunge

marcelstoer, 2021-12-29:
> Maybe just state the limitation to the documentation of the example.

Already there: https://github.com/nodemcu/nodemcu-firmware/blob/release/lua_examples/email/send_email_smtp.lua#L3

```lua
-- @description a basic SMTP email example. You must use an account which can
-- provide unencrypted authenticated access.
-- This example was tested with an AOL and Time Warner email accounts.
-- GMail does not offer unencrypted authenticated access.
```