nodemcu/nodemcu-firmware #3484: Error 'invalid user data' in net.socket:getpeer()

wangzexi, 2021-12-26:
Device: ESP8266

```lua
print('Setting up WIFI...')
wifi.setmode(wifi.STATION)
wifi.sta.config({ssid = '401_403', pwd = 'keyanlou1401'})

iptmr = tmr.create()
iptmr:register(1000, tmr.ALARM_AUTO, function()
    if wifi.sta.getip() == nil then
        print('Waiting for IP ...')
        return
    end
    print('IP is ' .. wifi.sta.getip())
    iptmr:unregister()
end)
iptmr:start()

srv = net.createServer()
srv:listen(80, function(conn)
    print(conn.getpeer()) -- 💣
    print(conn.getaddr())
end)
```

```
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: release
	commit: d4ae3c364bd8ae3ded8b77d35745b7f07879f5f9
	release: 
	release DTS: 202105102018
	SSL: false
	build type: float
	LFS: 0x0 bytes total capacity
	modules: file,gpio,net,node,sjson,tmr,uart,wifi
 build 2021-12-21 12:55 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
Setting up WIFI...
> Waiting for IP ...
Waiting for IP ...
Waiting for IP ...
Waiting for IP ...
IP is 192.168.1.101
PANIC: unprotected error in call to Lua API (init.lua:18: invalid user data)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 27444, room 16 
tail 4
chksum 0x03
load 0x3ffe8000, len 2564, room 4 
tail 0
chksum 0x9d
load 0x3ffe8a04, len 8, room 8 
tail 0
chksum 0xea
csum 0xea
```

pjsg, 2021-12-26:
I suspect it ought to be `conn:getpeer()`

wangzexi, 2021-12-27:
OMG, Thanks! I confuse the syntax of this call in Lua.

closed 2021-12-27 by wangzexi