nodemcu/nodemcu-firmware #3522: node.LFS.reload() LFS Image too big for configured LFS region

ENC-Automation, 2022-06-04:
### Expected behavior
Reload LFS with the flash image provided.
### Actual behavior
LFS Image too big for configured LFS region.

LFS: 0x16000 bytes (90112 bytes) total capacity
Image file size is 54927 bytes
54927 bytes should fit into 90112 bytes
### Test code
```Lua
print(node.LFS.reload("lfs.img"))
> LFS Image too big for configured LFS region
print("The LFS size is "..node.getpartitiontable().lfs_size)
> The LFS size is 90112
print("Image file size is "..file.stat("lfs.img").size)
> Image file size is 54927
print("Flash size is "..node.flashsize())
> Flash size is 4194304

do
  local s,p={},node.getpartitiontable()
  for _,k in ipairs{'lfs_addr','lfs_size','spiffs_addr','spiffs_size'} do
    s[#s+1] ='%s = 0x%06x' % {k, p[k]}
  end
  print ('{ %s }' % table.concat(s,', '))
end

> { lfs_addr = 0x07c000, lfs_size = 0x016000, spiffs_addr = 0x092000, spiffs_size = 0x36b000 }
```
### NodeMCU startup banner
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: release
	commit: f25dc56d3c6213b8ac7ce46d1293466137746eae
	release: 
	release DTS: 202112300746
	SSL: false
	build type: float
	LFS: 0x15000 bytes total capacity
	modules: adc,dht,encoder,file,gpio,i2c,mqtt,net,node,rtctime,sntp,tmr,u8g2,uart,wifi
 build 2022-06-04 00:47 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
### Hardware
Nodemcu Amica (ESP-12F) Silicon Labs CP2102

ENC-Automation, 2022-06-06:
Thank you HHHartman,

Sorry about my english.

Total *.lua applications files: 78Kb

Using -m luac.cross switch to limit size:

luac.cross -o lfs.img **-m 98304** -f *.lua
_luac.cross: The image is too large for specfied LFS size_
luac.cross -o lfs.img **-m 131072** -f *.lua
_luac.cross: The image is too large for specfied LFS size_
luac.cross -o lfs.img **-m 151552** -f *.lua  (148Kb)
_Ok. lfs.img file size is **59929** bytes._

It was necessary to set a size 2 times larger to generate the image.
So the LFS region had to be configured to be 2 times larger in order to work.
So Lua applications with up to 256Kb are not possible. At most something around 128Kb i believe.
Is there any future possibility to increase this "256Kb" of Lua applications?"
Thanks again for your time, dedication and work done.

HHHartmann, 2022-06-12:
again, the img file is compressed. The actual lfs image which is written to flash is then decompressed and thus matches the value given for the -m switch and the lfs region.

ENC-Automation, 2022-06-12:
Okay, HHHartmann, thanks again.

After testing, I came to the following conclusion:

Setting the maximum LFS region to:  LFS: 0x40000 bytes total capacity (256Kbytes)
Maximum Lua applications will be **128Kbytes.** _(After code editor, Before compilation with luac.cross)_
luac.cross -m switch it's irrelevant.
Therefore, the documentation should include:

_LFS support
...This now enables NodeMCU developers to create Lua applications with up to **128Kb** (Not 256Kb)  Lua code and read-only constants executing out of flash. All of the RAM is available for read-write data!_

Finally, I would like to thank all the developers and contributors of nodemcu-firmware which is of great value to me.