nodemcu/nodemcu-firmware #3526: Support the USB CDC device in ESP32-C3

pjsg, 2022-06-24:
### Missing feature
If you are using the USB CDC device in the ESP32-C3 for the console, then it doesn't work. This feature would add support for using the CDC device as the console. Ideally, it would also appear as a uart so that the `uart` family of functions would also work. However, that raises the issue of what *number* the CDC device should be. The other two uart devices also exist. Either we use 0 for the CDC device (if it is configured) and push the real uarts down to 1 & 2, or we do something else. Opinions wanted.

Yes, I will implement this.

### Justification
It makes some cheap boards usable.

### Workarounds
You have to use an external Serial-USB chip as a workaround.

tomsci, 2022-10-09:
Does this issue also apply to the S2 series? Or is USB done differently again on the C3? I'm trying get the Lua REPL to work over the native USB-C port (on an Adafruit feather ESP32-S2 board) using the `dev-esp32-idf4` branch and I haven't figured out yet if I'm just hitting configuration issues or there's something actually missing.

I've configured `Component config -> ESP System Settings -> Channel for console output = USB CDC` (ie `ESP_CONSOLE_USB_CDC`) and enabled TinyUSB and `TINYUSB_CDC_ENABLED` and while that did result in a usbmodem tty device enumerating, nodemcu isn't using it as far as I can see? Thanks to the annoyance of native USB and without a REPL I can't immediately check if console output is actually going there or not. Would REPL support require some tinyusb glue code to be added to platform.c, or similar?

pjsg, 2022-10-09:
I would hope that this code would work on your platform as well. Can you
try it and let me know?

tomsci, 2022-10-10:
I looked at #3536 but I'm not clear on whether USB serial JTAG console is what I want, or if the S2 even supports it. `ESP_CONSOLE_USB_SERIAL_JTAG` depends on `IDF_TARGET_ESP32C3 || IDF_TARGET_ESP32S3` in the Kconfig so presumably it doesn't? `ESP_CONSOLE_USB_SERIAL_JTAG` is a different thing to `ESP_CONSOLE_USB_CDC` right? Since I'm not wanting to do anything with USB JTAG and just want a USB serial port I can talk to, presumably that rules out `ESP_CONSOLE_USB_SERIAL_JTAG` as not the option for me?

pjsg, 2022-10-10:
It is entirely possible that it is just a question of fiddling with the preconditions and the #if statements to have it work for the S2. I don't have any S2 boards at the moment, so it is not possible for me to test it. However, I'm ordering a few, and I'll see what I can do.

tomsci, 2022-10-10:
Some more digging on https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/index.html (compared to the S3 docs which have a guide called [USB Serial/JTAG Controller Console](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/usb-serial-jtag-console.html)) it looks like the S2 doesn't have integrated USB Serial/JTAG Controller hardware, so I assume that means #3536 won't work for me. Sorry for the maybe obvious questions, still figuring out how this stuff works and what the hardware I have can do (impulse buy from Adafruit, what can I say). Now I think about it, I also have an S3 devboard which 2 USB-B micro ports on it, I guess one is connected to the dedicated serial/jtag controller, and the S2 board I'm working with here has only one USB-C port.

Sorry for the noise, I guess for now I'm out of luck with the S2 until someone hooks up a USB CDC impl using the tinyusb stack? I'd try it myself but clearly I don't know what I'm doing :D

pjsg, 2022-10-10:
I *think* that there is a config option in the menuconfig that provides a CDC console for the S2. This is the approach that I'm intending to start with (once my boards arrive in December!)

tomsci, 2022-10-10:
You're right, I was misunderstanding how tinyusb fitted in, and there is a CDC console provided via VFS without having to go full tinysusb. I swear I tried this option yesterday and it didn't seem to work, but I tried it just now and apparently it does.

Silly question, VFS in v4.4 seems to support using `/dev/console` as a read-write console which will talk to whichever of UART, USB JTAG or USB CDC is configured. Is there any reason why we couldn't just reimplement the Lua REPL in terms of that rather than having to have separate code for each in platform.c?

> once my boards arrive in December

Ouch. Supply chains suck.

tomsci, 2022-10-16:
After some banging of head against various walls, I got the Lua REPL working with the native USB CDC console on the S2/S3 based on @pjsg's usbcdc-only branch, with this diff: https://github.com/pjsg/nodemcu-firmware/compare/usbcdc-only...tomsci:nodemcu-firmware:tomsci-usbcdc-s2 .

So the answer to my question I think is "yes" - in theory I think we could dedup a bunch of code by making the uart, USB serial JTAG, and USB CDC console code paths all just use reads from the VFS `stdin`.

Annoyingly the CDC ACM aggressively buffers writes where it shouldn't (see [this comment](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_system/port/soc/esp32s2/usb_console.c#L328)) meaning I've had to hack in some `fsync()` calls to get the Lua REPL to work. And there are other issues with eg getting nodemcu-uploader to work (such as the USB stack [rebooting the board](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_system/port/soc/esp32s2/usb_console.c#L151) when the host disconnects the CDC), but it's a start.