nodemcu/nodemcu-firmware #3551: MQTT module fails to start on `dev-esp32-idf4'

KT819GM, 2022-10-16:
This is more of a request for a core devs to take a look into a MQTT module on the idf4 dev branch as in my point of view it's second to most important features for these microcontrollers. Built on Lua 5.3

### Expected behavior
MQTT module will work.
### Actual behavior
```
dofile("mqtt.lua")
[0;31mE (17078) MQTT_CLIENT: Error create mqtt task[0m
Lua error: 	mqtt.lua:10: Error starting mqtt client
stack traceback:
	[C]: in method 'connect'
	mqtt.lua:10: in main chunk
	[C]: in function 'ROM.dofile'
	stdin:1: in main chunk
	[C]: in ?
	[C]: in ?
```
### Example code part

```Lua
m = mqtt.Client("clientid", 120)

-- init mqtt client with logins, keepalive timer 120sec
m = mqtt.Client("clientid", 120, "user", "password")

-- setup Last Will and Testament (optional)
-- Broker will publish a message with qos = 0, retain = 0, data = "offline" 
-- to topic "/lwt" if client don't send keepalive packet
m:lwt("/lwt", "offline", 0, 0)
m:connect("192.168.11.118", 1883, 0, function(client)
  print("connected")

end)
```
### NodeMCU startup banner
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400257c1
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x2180
load:0x4004c000,len:0xd38
load:0x40050000,len:0x3868
entry 0x4004c29c
[0;32mI (26) boot: ESP-IDF v4.4.2 2nd stage bootloader[0m

### Hardware
esp32s2-devkit

serg3295, 2022-10-16:
MQTT module works for me. esp32-idf4 branch, latest commit.
```lua
user = "***"
password = "***"
broker = "192.168.1***"

wifi.start()
wifi.sta.connect()
-- wait for wifi to start

m = mqtt.Client("clientid", 120)
m = mqtt.Client("clientid", 120, user, password)
m:lwt("/lwt", "offline", 0, 0)
m:connect(broker, 1883, 0, function(client)
  print("connected")
end)
```
[image]

### Hardware
ESP32 Devkitc_v4

### NodeMCU startup banner
```
I (462040) wifi:state: run -> init (0)
I (462040) wifi:pm stop, total sleep time: 295903119 us / 451146252 us

W (462040) wifi:<ba-del>idx
I (462040) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (462110) wifi:flush txq
I (462110) wifi:stop sw txq
I (462110) wifi:lmac stop hw txq
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
entry 0x40080694
[0;32mI (29) boot: ESP-IDF v4.4.2 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 11:23:37[0m
[0;32mI (29) boot: chip revision: 1[0m
[0;32mI (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (39) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (43) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (52) boot: Enabling RNG early entropy source...[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00180000[0m
[0;32mI (91) boot:  3 lfs              unknown          c2 01 00190000 00010000[0m
[0;32mI (99) boot:  4 storage          Unknown data     01 82 001a0000 00070000[0m
[0;32mI (106) boot: End of partition table[0m
[0;32mI (110) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (118) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2312ch (143660) map[0m
[0;32mI (178) esp_image: segment 1: paddr=00033154 vaddr=3ffb0000 size=03728h ( 14120) load[0m
[0;32mI (184) esp_image: segment 2: paddr=00036884 vaddr=40080000 size=09794h ( 38804) load[0m
[0;32mI (200) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=b73d0h (750544) map[0m
[0;32mI (472) esp_image: segment 4: paddr=000f73f8 vaddr=40089794 size=0c380h ( 50048) load[0m
[0;32mI (493) esp_image: segment 5: paddr=00103780 vaddr=400c0000 size=00064h (   100) load[0m
[0;32mI (493) esp_image: segment 6: paddr=001037ec vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (509) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (509) boot: Disabling RNG early entropy source...[0m
[0;32mI (521) cpu_start: Pro cpu up.[0m
[0;32mI (522) cpu_start: Starting app cpu, entry point is 0x4008144c[0m
[0;32mI (508) cpu_start: App cpu up.[0m
[0;32mI (538) cpu_start: Pro cpu start user code[0m
[0;32mI (538) cpu_start: cpu freq: 160000000[0m
[0;32mI (538) cpu_start: Application information:[0m
[0;32mI (543) cpu_start: Project name:     nodemcu[0m
[0;32mI (548) cpu_start: App version:      1.4.0-master_20151229-857-gcad1[0m
[0;32mI (555) cpu_start: Compile time:     Oct 16 2022 11:23:33[0m
[0;32mI (561) cpu_start: ELF file SHA256:  ebcb6dd79561cc25...[0m
[0;32mI (567) cpu_start: ESP-IDF:          v4.4.2[0m
[0;32mI (572) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (579) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (585) heap_init: At 3FFB78D8 len 00028728 (161 KiB): DRAM[0m
[0;32mI (591) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (597) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (604) heap_init: At 40095B14 len 0000A4EC (41 KiB): IRAM[0m
[0;32mI (611) spi_flash: detected chip: generic[0m
[0;32mI (615) spi_flash: flash io: dio[0m
[0;32mI (620) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (630) uart: queue free spaces: 3[0m

LFS image loaded
I (750) wifi:wifi driver task: 3ffc7674, prio:23, stack:6656, core=0
[0;32mI (750) system_api: Base MAC address is not set[0m
[0;32mI (750) system_api: read default base MAC address from EFUSE[0m
I (770) wifi:wifi firmware version: eeaa27d
I (770) wifi:wifi certification version: v7.0
I (770) wifi:config NVS flash: enabled
I (770) wifi:config nano formating: disabled
I (780) wifi:Init data frame dynamic rx buffer num: 32
I (780) wifi:Init management frame dynamic rx buffer num: 32
I (790) wifi:Init management short buffer num: 32
I (790) wifi:Init dynamic tx buffer num: 32
I (800) wifi:Init static rx buffer size: 1600
I (800) wifi:Init static rx buffer num: 10
I (800) wifi:Init dynamic rx buffer num: 32
[0;32mI (810) wifi_init: rx ba win: 6[0m
[0;32mI (810) wifi_init: tcpip mbox: 32[0m
[0;32mI (820) wifi_init: udp mbox: 6[0m
[0;32mI (820) wifi_init: tcp mbox: 6[0m
[0;32mI (820) wifi_init: tcp tx win: 5744[0m
[0;32mI (830) wifi_init: tcp rx win: 5744[0m
[0;32mI (830) wifi_init: tcp mss: 1436[0m
[0;32mI (840) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (840) wifi_init: WiFi RX IRAM OP enabled[0m

NodeMCU ESP32 build unspecified powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v4.4.2
cannot open init.lua: No such file or directory
```

KT819GM, 2022-10-16:
Thank you for a check. Seems something will be related to my `menuconfig` then. Will try to find what is causing it and then will close the issue.

KT819GM, 2022-10-31:
Wasn't able to find the root cause but it was definitely related to 'menuconfig' options. Will not hold this "issue" active anymore as with default settings mqtt is working.

closed 2022-10-31 by KT819GM