nodemcu/nodemcu-firmware #3553: Support for AHT20 sensor (lua)

MudassirTufail, 2022-10-30:
### Missing feature
I do not see lua modules to support aht20 (temperature, humidity) sensor on nodemcu 

### Justification
With BME sensor disappearing from the market, an alternate option is based on combo sensor (BMP280+AHT20) - please refer to https://universal-solder.ca/product/aht20-bmp280-sensor-module-replaces-bme280/
To make this work with nodemcu, we currently are missing lua module for aht20.

### Workarounds
workarounds are more expensive

vsky279, 2022-11-05:
Hi,
Have you tried to get inspiration from the [bme280.lua](https://github.com/nodemcu/nodemcu-firmware/blob/f25dc56d3c6213b8ac7ce46d1293466137746eae/lua_modules/bme280/bme280.lua) lua module?
It should not be that difficult to write a new lua module for AHT20.
I don't have this piece of hardware with me but it should communicate via I2C on address 0x38.
I will order 1 anyway.

marcelstoer, 2022-11-05:
See #1010

closed 2022-11-05 by marcelstoer