nodemcu/nodemcu-firmware #3555: node.LFS.reload requires compressed lfs.img

weskerfoot, 2022-11-09:
### Expected behavior

According to the docs, this should result in a usable lfs.img
`luac.cross.int -o lfs.img *lua`

Which I can then load using `node.LFS.reload("lfs.img")`

### Actual behavior

The firmware requires it to be in a different format (for `uzlib_inflate`)

```
  /* first inflate pass */
  if (uzlib_inflate (get_byte, put_byte, recall_byte,
                     in->len, &crc, &in->inflate_state) < 0)
    flash_error("read error on LFS image file");

```
I get this error (from this invocation) if I try running `print(node.LFS.reload("lfs.img"))` after uploading using `nodemcu-tool upload lfs.img` 

If I do `luac.cross.int -m 20480 -o lfs.img *lua` it works just fine and restarts as expected, because of the different format I assume.

```
[wes@stark sources]$ /home/wes/code/nodemcu-firmware/luac.cross.int -o lfs.img *lua
[wes@stark sources]$ file lfs.img
lfs.img: Lua bytecode, version 5.1
[wes@stark sources]$ /home/wes/code/nodemcu-firmware/luac.cross.int -m 20480 -o lfs.img *lua
[wes@stark sources]$ file lfs.img
lfs.img: gzip compressed data, max speed, from Unix, original size modulo 2^32 19412
```

### NodeMCU startup banner
```
NodeMCU 3.0.0.0 wes
        branch: release
        commit: f25dc56d3c6213b8ac7ce46d1293466137746eae
        release: 3.0.0-release_20211229 +1
        release DTS: 202112300746
        SSL: false
        build type: integer
        LFS: 0x40000 bytes total capacity
        modules: enduser_setup,file,gpio,net,node,pwm,rtctime,sntp,tmr,uart,wifi
 build 2022-11-09 18:26 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```

NodeMCU-Tool version is `3.2.1`

### Hardware
Tested using https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/pinouts

fikin, 2022-11-10:
against current dev branch, using build options LUA=53,
```
$ ./luac.cross -m 20480 -o local/fs/aa.img local/lua/*.lua
$ ./luac.cross -o local/fs/bb.img local/lua/*.lua
$ file local/fs/aa.img local/fs/bb.img
local/fs/aa.img: Lua bytecode, version 5.3
local/fs/bb.img: Lua bytecode, version 5.3
```
startup banner:
```
NodeMCU 3.0.0.0 
	branch: lua53
	commit: fe31d6c0881811557da3869a8fae244342ee0657
	release: 3.0.0-release_20211229 +12
	release DTS: 202211082316
	SSL: false
	build type: float
	LFS: 0x20000 bytes total capacity
	modules: adc,bit,enduser_setup,file,gpio,i2c,mqtt,net,node,ow,rtcmem,rtctime,sjson,sntp,spi,tmr,uart,wifi
 build 2022-11-09 02:03 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)
```

HHHartmann, 2022-11-13:
I am using that scenario every day and it works.
You obviously are doing "int" and "regular" builds. each build also creates a matching luac.cross/luac.cross.int (They also create different bytecode)

I assume that your luac.cross is outdated as the compression feature was added in a second step. Just make sure to use the correct luac.cross that came with the FW build and it should be fine.

btw the -m option is not implemented for Lua 5.3 (it does nothing)

weskerfoot, 2022-11-14:
> 

I made sure to compile everything fresh and was still getting that error message `"read error on LFS image file"` until I used the `-m` flag which resulted in a gzip file. I used the integer build, but the firmware did match the cross compiler as well, so I don't see how that would have made a difference. Anyway, it works for my use case so I can continue using it as is.

weskerfoot, 2022-11-14:
I see the problem now, I forgot to pass the `-f` flag, but for some reason it still works if you just pass `-m` which gives you a gzip file. Would it be acceptable if I changed that error message to be a bit clearer? E.g. "read error on LFS image file. Make sure you passed the -f flag" or something?