nodemcu/nodemcu-firmware #3559: sjson.NULL is missing

fikin, 2022-11-20:
### Expected behavior
Decoded null values are represented as "userdata: (nil)", which according doc should be identical to sjson.NULL constant.

### Actual behavior
it appears sjson.NULL is missing completely from module exports ;)

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
> =sjson.decode('[null]')
table: 0x3fff5040
> =sjson.decode('[null]')[1]
userdata: (nil)
> =sjson.decode('[null]')[1]==sjson.NULL
false
> =type(sjson.NULL)
nil
```
### NodeMCU startup banner
NodeMCU 3.0.0.0 
	branch: lua53
	commit: 6fdfc48eeea5da86db22d2e469cc9d8885f1ae69
	release: 3.0.0-release_20211229 +12
	release DTS: 202211122042
	SSL: false
	build type: float
	LFS: 0x20000 bytes total capacity
	modules: adc,bit,crypto,encoder,file,gpio,i2c,net,node,rtcmem,rtctime,sjson,sntp,spi,tmr,uart,wifi
build 2022-11-20 15:14 powered by Lua 5.3.5 on SDK 3.0.1-dev(fce080e)

### Hardware
WeMos D1

marcelstoer, 2022-12-06:
@pjsg I vaguely remember you fixed something like this in the past, correct?

fikin, 2023-03-03:
@pjsg : i wonder would you like to give me some implementation suggestions and i can add a NULL constant to the module?

form what i've seen so far, the workaround is to json decode '{"a":null}' and use that. but there should be some constant already in the code to act as same, isn't there?

pjsg, 2023-03-13:
Hmm. The problem is that the `null` value is actually on a per encoder basis. This feels broken to me and I think that there ought to be a `sjson.NULL` that has a specific value. I suspect that you need a row like    `LROT_LUDENTRY(NULL, JSON_NULL)` in the block that begins `LROT_BEGIN(sjson, NULL, 0)`. But I'm not quite sure how the `JSON_NULL` should be defined.