nodemcu/nodemcu-firmware #3565: Problems with wifi module on ESP32

KSDaemon, 2023-01-01:
Hi there! Sorry, maybe I'm missing something, but can't figure out... Hope you can help me.
I dug through the docs, and tried to search — but no luck...

### Expected behavior

Wifi module functions work as expected/described in docs

### Actual behavior

Most of the wifi module functions a nils...

```
=wifi.setmode(wifi.STATION)
stdin:1: attempt to call field 'setmode' (a nil value)

=wifi.eventmon.register
stdin:1: attempt to index field 'eventmon' (a nil value)
```

### Test code
Just doing the basic code examples from the [docs](https://nodemcu.readthedocs.io/en/release/upload/): trying to connect to wifi from init.lua script. But hitting `nils` in most cases, like: 

```Lua
=wifi.setmode(wifi.STATION)
-- or 
=wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, wifi_connect_event)
```
### NodeMCU startup banner
```
NodeMCU ESP32 built on nodemcu-build.com provided by frightanic.com
	branch: dev-esp32
	commit: eb56f6237ba00c69b4f4e270b8ef4e3fe1d45982
	SSL: true
	modules: crypto,encoder,file,gpio,http,mqtt,net,node,sjson,sodium,tmr,uart,wifi
 build 2023-01-01-21-02-54 powered by Lua 5.1.4 on ESP-IDF v3.3-beta1-1738-gb64b375234 on SDK IDF
```

### Hardware
ESP32 DEVKIT

serg3295, 2023-01-02:
You should use [this docs](https://nodemcu.readthedocs.io/en/dev-esp32/modules/wifi/) for esp32.

KSDaemon, 2023-01-02:
Oh! Thnx @serg3295 For pointing to docs!

KSDaemon, 2023-01-02:
I'll close issue then.

closed 2023-01-02 by KSDaemon