nodemcu/nodemcu-firmware #3570: Firmware not working on ESP32-S2 mini

carlo161, 2023-02-13:
### Expected behavior
Normal behavior of a firmware on ESP32 board.

### Actual behavior
The board apparently does not boot anymore.

### Test code

### NodeMCU startup banner
Last firmware version ([3.0.0-release_20211229](https://github.com/nodemcu/nodemcu-firmware/releases/tag/3.0.0-release_20211229))

### Hardware
ESP32-S2 mini [https://www.wemos.cc/en/latest/s2/s2_mini.html](https://www.wemos.cc/en/latest/s2/s2_mini.html)

I tried to build and flash the firmware on the board (I used many ESP in the past but never this newer version) and I am not able of replicating the same toolchain on it.
Is the firmware not compatible with it?

Plaque-fcc, 2023-07-20:
As of today, switching to ESP-S2 by set-target esp32s2, building the firmware and flashing it, I get my ESP32-S2 mini in an unusable state when it doesn't provide UART over USB.

Plaque-fcc, 2023-07-20:
I bet #3526 is related to this.
Looks like #3536 is going to fix it.