nodemcu/nodemcu-firmware #3603: ESP32S3: Firmware startup but no lua prompt

ThomasKaemmerer, 2023-07-05:
### Expected behavior
Active lua prompt.

### Actual behavior
I use the current state of the firmware sources to create the necessary images for this device. After successfully flashing and rebooting the device, the FW is loaded but not completely executed. No lua prompt appears and no further interaction with the device is possible.

### Test code
How I would love to try out a lua snippet ;-)

### NodeMCU startup banner
```
79) esp_image: segment 1 map_segment 0 segment_data_offs 0x35a2cet paddr=00040000 vaddr=42000000 size=748712 n=12
V (785) boot: rc=0
D (786) boot: start: 0x40375664
I (800) cpu_start: Pro cpu up.
I (801) cpu_start: Starting app cpu, entry point is 0x403755c0
I (0) cpu_start: App cpu up.
I (812) cpu_start: Pro cpu start user code
I (812) cpu_start: cpu freq: 240000000
I (812) cpu_start: Application information:
I (812) cpu_start: Project name:     nodemcu
I (816) cpu_start: App version:      1.4.0-master_20151229-873-gf125
I (822) cpu_start: Compile time:     Jul  5 2023 11:36:28
I (827) cpu_start: ELF file SHA256:  348c06f024d335d9...
I (832) cpu_start: ESP-IDF:          v4.4.3
I (836) heap_init: Initializing. RAM available for dynamic allocation:
I (842) heap_init: At 3FCA0D10 len 00048A00 (290 KiB): D/IRAM
I (847) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (853) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (858) heap_init: At 600FE02C len 00001FD4 (7 KiB): RTCRAM
I (865) spi_flash: detected chip: winbond
I (868) spi_flash: flash io: dio
I (871) sleep: Configure to isolate all GPIO pins in sleep state
I (876) sleep: Enable automatic switching of GPIO sleep configuration
I (883) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (892) uart: queue free spaces: 3

No LFS image loaded
I (952) pp: pp rom version: e7ae62f
I (952) net80211: net80211 rom version: e7ae62f
I (962) wifi:wifi driver task: 3fca3b18, prio:23, stack:6656, core=0
I (962) system_api: Base MAC address is not set
I (962) system_api: read default base MAC address from EFUSE
I (972) wifi:wifi firmware version: 8cb87ff
I (972) wifi:wifi certification version: v7.0
I (972) wifi:config NVS flash: enabled
I (972) wifi:config nano formating: disabled
I (982) wifi:Init data frame dynamic rx buffer num: 32
I (982) wifi:Init management frame dynamic rx buffer num: 32
I (992) wifi:Init management short buffer num: 32
I (992) wifi:Init dynamic tx buffer num: 32
I (1002) wifi:Init static tx FG buffer num: 2
I (1002) wifi:Init static rx buffer size: 1600
I (1002) wifi:Init static rx buffer num: 10
I (1012) wifi:Init dynamic rx buffer num: 32
I (1012) wifi_init: rx ba win: 6
I (1012) wifi_init: tcpip mbox: 32
I (1022) wifi_init: udp mbox: 6
I (1022) wifi_init: tcp mbox: 6
I (1022) wifi_init: tcp tx win: 5744
I (1032) wifi_init: tcp rx win: 5744
I (1032) wifi_init: tcp mss: 1440
I (1032) wifi_init: WiFi IRAM OP enabled
I (1042) wifi_init: WiFi RX IRAM OP enabled
No LFS image loaded

NodeMCU ESP32 build unspecified powered by Lua 5.1.4 [5.1-integral] on IDF v4.4.3
cannot open init.lua: No such file or directory
```

### Hardware
LILYGO T-Display S3 AMOLED

Plaque-fcc, 2023-07-20:
I suppose, it expects you to upload some init.lua to the board for it to execute automatically.

marcelstoer, 2024-06-24:
> open init.lua: No such file or directory

You either upload/execute code ad-hoc with something like ESPlorer or upload an `init.lua`.

closed 2024-06-24 by marcelstoer