nodemcu/nodemcu-firmware #3614: Release version issue snprintf

vit1251, 2023-10-15:
### Expected behavior

DEPEND: CC app/libc/snprintf.c
snprintf.c:41:35: fatal error: c99-snprintf/snprintf.c: No such file or directory
 #include "c99-snprintf/snprintf.c"
                                   ^
compilation terminated.
CC app/libc/snprintf.c
snprintf.c:41:35: fatal error: c99-snprintf/snprintf.c: No such file or directory
 #include "c99-snprintf/snprintf.c"
                                   ^
compilation terminated.

th3worst4, 2025-01-10:
I've cloned the snprintf.c file from this repository https://github.com/weiss/c99-snprintf/blob/master/snprintf.c at the required folder. It solved the "No such file or directory" but now I'm having linking errors 
```
/home/caioc/nodemcu-firmware/sdk/esp_iot_sdk_v3.0-e4434aa/lib/libc.a(lib_a-svfprintf.o):(.literal+0x10): undefined reference to `_malloc_r'
```

vit1251, 2025-01-23:
Look like you get `snprintf` implementation work with heap. I acutally does not know ESP NO-OS is provide `malloc`, `free` on heap inside. It may be need to implement or use `sprintf` without heap allocation.

In any case look like 2 years question stil not answered and I think project dead.