nodemcu/nodemcu-firmware #3620: Flashing on ESP32-C2 not working

ceifa, 2024-01-10:
### Expected behavior
`Failed to load init.lua`

### Actual behavior
Flashing output:
```
esptool.py v4.7.0
Found 2 serial ports
Serial port COM4
Connecting.....
Detecting chip type... ESP32-C2
Chip is ESP32-C2 (revision v1.0)
Features: WiFi, BLE
Crystal is 26MHz
MAC: 34:98:7a:0f:2d:58
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Unexpected chip id in image. Expected 12 but value was 27444. Is this image for a different chip model?
Flash will be erased from 0x00000000 to 0x00091fff...
Compressed 598016 bytes to 398328...
Wrote 598016 bytes (398328 compressed) at 0x00000000 in 35.2 seconds (effective 135.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Serial terminal output:
```
d�r��l�{�l���ccp�n��
scl�l������d��l`?�l�l{$b�o��8o��n��lg�b$��{���d�
��������c��gn�og��c��o|��|�g��o���d`s��g�c�ocp��$
l`
```

### Test code
```
esptool.exe write_flash 0x00000 nodemcu-release-5-modules-2024-01-08-15-14-34-integer.bin
```

### Hardware
ESP32-C2

pjsg, 2024-02-02:
How did you build the image?

ceifa, 2024-02-02:
Cloud builder, with no special configs.
[image]

marcelstoer, 2024-02-03:
Unfortunately the cloud builder can't build for ESP32 (anymore).

closed 2024-02-03 by marcelstoer

ceifa, 2024-02-06:
Why this limitation? I'm very newbie on IOT, cloud build can help a lot!