nodemcu/nodemcu-firmware #3657: ESP32: uart.write() doesn't send long chunk via CONSOLE_USB_SERIAL_JTAG

serg3295, 2024-07-30:
### Expected behavior
Currently, to send a data fragment larger than 256 bytes via CONSOLE_USB_SERIAL_JTAG, a forced insertion of `\n` between 256-byte fragments is required.

I would like the behavior to match the behavior when sending a long sequence via UART_DEFAULT. That is, data should be sent continuously without adding `\n`

In the nodemcu-tool and nodemcu-tools utilities, uniterrupted sending is used to download a file from the ESP32.
Forcing `\n` to each piece will break the exchange protocol.

Moreover, the size of the response to a request from the utility can exceed 256 bytes. And in this case, I can't even add `\n`

### Actual behavior
When sending a chunk longer than 256 bytes, only the first 256 bytes are sent.
The remaining bytes will be sent only if you add `\n` at the end of the chunk.
### Test code

This code works
```Lua
s = string.rep('a', 256)
uart.write(0, s)
uart.write(0,'\n')
uart.write(0, s)
print'\n'
```
This code does not work
```Lua
s = string.rep('a', 256)
uart.write(0, s)
uart.write(0, s)
print'\n'
```
### NodeMCU startup banner
<details>
  <summary>Startup banner</summary>
  
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1704
load:0x403cc710,len:0x938
load:0x403ce710,len:0x2ee8
entry 0x403cc710
I (24) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (24) boot: compile time Jul 29 2024 20:30:34
I (24) boot: chip revision: v0.4
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (32) boot.esp32c3: SPI Mode       : DIO
I (36) boot.esp32c3: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00180000
I (80) boot:  3 lfs              unknown          c2 01 00190000 00010000
I (87) boot:  4 storage          Unknown data     01 82 001a0000 00070000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=2d748h (186184) map
I (137) esp_image: segment 1: paddr=0003d770 vaddr=3fc92400 size=028a8h ( 10408) load
I (139) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=d19b8h (858552) map
I (279) esp_image: segment 3: paddr=001119e0 vaddr=3fc94ca8 size=001ach (   428) load
I (280) esp_image: segment 4: paddr=00111b94 vaddr=40380000 size=1237ch ( 74620) load
I (299) esp_image: segment 5: paddr=00123f18 vaddr=50000000 size=00018h (    24) load
I (305) boot: Loaded app from partition at offset 0x10000
I (306) boot: Disabling RNG early entropy source...
I (308) cpu_start: Unicore app
I (312) cpu_start: Pro cpu up.
I (327) cpu_start: Pro cpu start user code
I (327) cpu_start: cpu freq: 160000000 Hz
I (327) cpu_start: Application information:
I (330) cpu_start: Project name:     nodemcu
I (335) cpu_start: App version:      tmr-libmain-binpatch150-888-gca
I (342) cpu_start: Compile time:     Jul 29 2024 20:32:43
I (348) cpu_start: ELF file SHA256:  b3501ddbd7970a59...
I (354) cpu_start: ESP-IDF:          v5.1.3
I (359) cpu_start: Min chip rev:     v0.3
I (364) cpu_start: Max chip rev:     v1.99 
I (369) cpu_start: Chip rev:         v0.4
I (373) heap_init: Initializing. RAM available for dynamic allocation:
I (381) heap_init: At 3FC997B0 len 00026850 (154 KiB): DRAM
I (387) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (394) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (402) heap_init: At 50000018 len 00001FD0 (7 KiB): RTCRAM
I (409) spi_flash: detected chip: generic
I (412) spi_flash: flash io: dio
W (417) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (427) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (436) sleep: Configure to isolate all GPIO pins in sleep state
I (442) sleep: Enable automatic switching of GPIO sleep configuration
I (449) app_start: Starting scheduler on CPU0
I (454) main_task: Started on CPU0
I (454) main_task: Calling app_main()

E (464) mmap: esp_mmu_paddr_to_vaddr(752): paddr isn't mapped
LFS image loaded
I (474) pp: pp rom version: 9387209
I (474) net80211: net80211 rom version: 9387209
I (484) wifi:wifi driver task: 3fca87a4, prio:23, stack:6656, core=0
I (484) wifi:wifi firmware version: 0016c4d
I (484) wifi:wifi certification version: v7.0
I (484) wifi:config NVS flash: enabled
I (484) wifi:config nano formating: disabled
I (484) wifi:Init data frame dynamic rx buffer num: 32
I (484) wifi:Init static rx mgmt buffer num: 5
I (484) wifi:Init management short buffer num: 32
I (484) wifi:Init dynamic tx buffer num: 32
I (484) wifi:Init static tx FG buffer num: 2
I (484) wifi:Init static rx buffer size: 1600
I (484) wifi:Init static rx buffer num: 10
I (484) wifi:Init dynamic rx buffer num: 32
I (484) wifi_init: rx ba win: 6
I (494) wifi_init: tcpip mbox: 32
I (494) wifi_init: udp mbox: 6
I (494) wifi_init: tcp mbox: 6
I (494) wifi_init: tcp tx win: 5760
I (494) wifi_init: tcp rx win: 5760
I (494) wifi_init: tcp mss: 1440
I (494) wifi_init: WiFi IRAM OP enabled
I (494) wifi_init: WiFi RX IRAM OP enabled

NodeMCU ESP32 build 2024-07-29 20:32 powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v5.1.3
cannot open init.lua: No such file or directory
  
</details>

### Hardware
esp32c3 and esp32c6 via CONSOLE_USB_SERIAL_JTAG

serg3295, 2024-08-08:
I found a workaround to the issue.
1. Upgrade to ifd v5.2.2. This is a mandatory step.
2. Change platform.c a little

```c++
void platform_uart_send_multi( unsigned id, const char *data, size_t len )
{
  size_t i;
  // if (id == CONFIG_ESP_CONSOLE_UART_NUM) {
      for( i = 0; i < len; i ++ ) {
        putchar (data[ i ]);
    }
  // } else {
    // uart_write_bytes(id, data, len);
  // }
}

void platform_uart_send( unsigned id, uint8_t data )
{
  // if (id == CONFIG_ESP_CONSOLE_UART_NUM)
    putchar (data);
  // else
    // uart_write_bytes(id, (const char *)&data, 1);
}

void platform_uart_flush( unsigned id )
{
  // if (id == CONFIG_ESP_CONSOLE_UART_NUM)
    fflush (stdout);
  // else
    // uart_tx_flush(id);
}
```

I don't know why it works, but it does.

jmattsson, 2024-10-20:
The underlying issue here is that the USB-SERIAL-JTAG isn't an actual UART in the first place. In the name of backwards compatibility I've tried to hide that fact with increasing amounts of special casing in the platform code. As the IDF evolves further towards POSIXness, and the introduction of CDC-ACM consoles as well, this approach has, frankly, gotten rotten. Not only has it produced inconsistent behaviour, odd edge cases (like this), but also clearly a bunch of confusion :(

To resolve all these issues I think we have to bite the bullet and get around to moving the console device access into its own module. We've avoided it because it's a backwards compatibility breaking change, but it really looks to be the lesser evil by now. Let me see if I can cook up a PR for evaluation today.

serg3295, 2024-10-21:
I have already rewritten the code of both utilities so that they work with the esp jtag console (on idf5.3.1) without using `uart.on`, `uart.start`, `uart.stop`. :-)

However, I was unable to implement sending binary data from esp to the host, in particular file dowload.
`uart.write` is the only thing that is now necessary for the utilities to work correctly with the firmware.

jmattsson, 2024-10-22:
I've only just started the testing, but if you want to have a play early the branch is [here](https://github.com/DiUS/nodemcu-firmware/tree/refs/heads/dev-esp32-console-module).

Effectively, the console interaction is now via the `console` module, and you have `console.on("data", ...)` available similar to what's on the `uart` module. There's also `console.mode(mode)` to control whether console data goes to the LVM automatically or not. Hopefully the [docs](https://github.com/DiUS/nodemcu-firmware/blob/refs/heads/dev-esp32-console-module/docs/modules/console.md) are good enough to describe it.

serg3295, 2024-10-22:
Thanks, I'll test that.

serg3295, 2024-10-22:
I tested the new firmware on esp32.
I use `uart.on("data", bufSize, function(d) <d is binary data to be written to a file> end, 0)` for uploading file to esp. 

Using new function console.mode(0) to disable passing data to Lua VM, I cannot get back to Interactive console.
Is there any way to get binary data in the console.on and then return to interactive mode?

jmattsson, 2024-10-23:
I've added an [example](https://github.com/DiUS/nodemcu-firmware/blob/dev-esp32-console-module/docs/modules/console.md#consolemode) on how to cleanly transfer potentially binary data using the `console` module, via a classic `STX`/`ETX`/`DLE` framing scheme.

jmattsson, 2024-10-23:
In regard to the original issue, I've also tested writing a 300 byte long string using `console.write()` and received that fine on a USB-Serial-JTAG console.

serg3295, 2024-10-23:
Thank you for such a detailed example! Now I figured it out.

Minor fixes in uart.md examples:

-- when 4 chars is received.
uart.on("data", 4,
  ...
end\~~,0\~~)
-- when '\r' is received.
uart.on("data", "\r",
  ...
end\~~, 0\~~)

I have been testing the console module on esp32 -s3, -c3, -c6, (jtag) 
It works, but I notice errors in my utilities while downloading file.
What is max buffer size for _console.write_(<buf_size_long_string?>)?

-s2 (cdc)

The issue is common for both _idf5.3.1_ and _dev-console_ branches. Very strange behavior. If the connection is made with esp reboot (`make monitor` or a terminal program HTerm 0.8.5), then REPL works.
When the connection is made without esp reboot (`npx @serialport/terminal -b115200 -p /dev/ttyACM0`    or my host utilities), there is no response to print'a'. Only node.restart() is accepted.
I will investigate this in more detail.

jmattsson, 2024-10-24:
Ah, thank for, I'll tidy up those uart examples!

There is no inherent limit to the string passed to `console.write()`, and I thought I did implement proper retry  logic on the writes. While much of the time I can write a 4097 byte string, sometimes it's silently cut off after 257 bytes here too. I'm looking into it now.

jmattsson, 2024-10-24:
Okay, I have a workaround that seems to do the trick reliably here on my c3. See if it works for your setup too @serg3295 ?

serg3295, 2024-10-24:
I am observing long chunks of ~4098 bytes from the ESP. Moreover, the UART console transmits data losslessly, and the  JTAG console loses data randomly starting with the second long data packet.
Perhaps, the IDF5.3.1 have UART errors been fixed?

jmattsson, 2024-10-24:
@serg3295 You're still seeing losses even after my most recent workaround?! It seems reliable now across UART, USB-Serial-JTAG and CDC-ACM here for me.

serg3295, 2024-10-24:
I apologize, I didn't pull the last changes. :-(

I have tested downloading large files from esp-s3. It's working now!
I will test it more thoroughly on other chips.

jmattsson, 2024-10-25:
I've just pushed `scripts/upload-file.py` to that branch, as a small utility both for direct use but also as a guide for how IDEs might want to use the new `console` module.

Plus I updated both the `uart` and `console` docs to make mention of it.

I think this is ready for a PR? The one other thing that would be lovely is if we could add something to our `lua_compat` that somehow wrangles legacy `uart.on()` (and whatever else IDEs have been sending) into something suitable for the `console` module, so that older IDEs can be used if you load said lua file.

serg3295, 2024-10-25:
I tested uploading and downloading files using `console.on`, `console.mode` and `console.write` and found no errors.

Also I reworked _nodemcu-tools_ VSCode extension and _NodeMCU-Tool_ CLI utility to support the new firmware. 
They now support all firmware versions, including the legacy `idf3` branch (with `file` module), the version that uses `uart.start/stop` and the latest `console` version.

jmattsson, 2024-10-25:
@serg3295 fabulous! I've raised the PR, let's see what the verdict is.

serg3295, 2024-10-25:
I have one more question.
ESP32-C3, jtag-console, uart module is included into firmware.
I tried calling `uart.setup` from _make monitor_ and got this response

```
NodeMCU ESP32 build 2024-10-25 08:40 powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v5.1.3
cannot open init.lua: No such file or directory
> uart.setup()
Lua error:      stdin:1: attempt to index a nil value (field 'uart')
stack traceback:
        stdin:1: in main chunk
        [C]: in ?
        [C]: in ?
> 
```

jmattsson, 2024-10-25:
It looks like the `uart` module isn't enabled/compiled in. Check your menuconfig under `Component config -> NodeMCU modules`

serg3295, 2024-10-25:
// sdkconfig
# CONFIG_NODEMCU_CMODULE_TIME is not set
CONFIG_NODEMCU_CMODULE_UART=y
# end of NodeMCU modules

serg3295, 2024-10-25:
However, no uart module in firmware
```
>   for k,v in pairs(getmetatable(_G).__index) do print(k.." : "..tostring(v)) end 
console : table: 0x3c0d6130
encoder : table: 0x3c0d621c
file : table: 0x3c0d646c
gpio : table: 0x3c0d6730
http : table: 0x3c0d6d10
i2c : table: 0x3c0d7060
net : table: 0x3c0d77f0
node : table: 0x3c0d81e0
ow : table: 0x3c0d84b8
pipe : table: 0x3c0d8584
tmr : table: 0x3c0d87cc
wifi : table: 0x3c0d8c60
string : table: 0x3c0dbbb8
table : table: 0x3c0dbd3c
coroutine : table: 0x3c0da930
debug : table: 0x3c0dacc8
math : table: 0x3c0db408
utf8 : table: 0x3c0dbe54
ROM : table: 0x3fc95a6c
assert : function: 0x4201e6ba
```

jmattsson, 2024-10-25:
What the... 🤯 Let me check here...

jmattsson, 2024-10-25:
Ugh, my bad 🤦‍♀️. Fixed now.
(I based the console module off the uart module, but then failed to fully update the module registration macro, leading the console module to claim to be the uart module. Remarkably silly mistake to make.)

serg3295, 2024-10-25:
Hooray, it worked!

serg3295, 2024-10-25:
I tested simultaneous connection via usb-jtag-console interface and via UART0 connected to FTDI adapter.

```mermaid
flowchart LR
host[["make monitor"]]
esp["ESP32-C3"]
ftdi[FTDI adapter]
terminal[[HTerm 0.8.5]]
host-- usb-jtag-console <-->esp<-- UART0 -->ftdi<-->terminal
```

Data is transferred in both directions from console to UART and vice versa.

```Lua
uart.setup(0, 115200, 8, uart.PARITY_NONE, uart.STOPBITS_1, {tx=21,rx=20})
uart.start(0)
uart.on(0,"data", "\n", function(data) print(data)end)
uart.write(0, "test string\n")
```

serg3295, 2024-10-25:
I'm trying to figure out the CDC console. I am unable to send a string larger than 89 bytes from the CLI utility.
If the string is exceeded 89 bytes, then I get only 89 bytes in the response and the exchange with esp freezes.

When sending a command from the utility's CLI, short commands are executed correctly, a command exceeding 89 bytes suspends the exchange.

```
>
[NodeMCU-Tool]~ Connected 
[device]      ~ Model: esp32s2 | Version: 0.0.undefined | ChipID: 0x_unknown | FlashID: 0x204016 
[serialport]  ~ TX: print(file.fsinfo())
[serialport]  ~ RX: pri
[serialport]  ~ RX: nt(fil
[serialport]  ~ RX: e.fsinfo())
[serialport]  ~ RX: 413899      502     414401
[serialport]  ~ RX: >
[serialport]  ~ TX: local l={}for k,v in pairs(file.list())do l[#l+1]=("%s:%d"):format(k,v)end;print(table.concat(l,";"))
[serialport]  ~ RX: lo
[serialport]  ~ RX: cal l=
[serialport]  ~ RX: {}for k,v in p
[serialport]  ~ RX: airs(file.list
[serialport]  ~ RX: ())do l
[serialport]  ~ RX: [#l+1]
[serialport]  ~ RX: =("%s:
[serialport]  ~ RX: %d"):
[serialport]  ~ RX: format
[serialport]  ~ RX: (k,v)
[serialport]  ~ RX: end;print(table
[serialport]  ~ RX: .conca

We're stuck here
```

Having tested the example `upload-file.py` I found that the maximum size of the uploaded file also cannot exceed 89 bytes. If the file size is exceeded 89 bytes, a zero size file is written to disk.
I haven't looked at the receiver function code yet, but I guess the reason is not in the code `upload-file.py`

file: `a.txt` ; length=89 byte
```
12345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
```

the file was successfully written
```
I (344) wifi_init: WiFi RX IRAM OP enabled
> local l={}for k,v in pairs(file.list())do l[#l+1]=("%s:%d"):format(k,v)end;print(table.concat(l,";"))
a.txt:89
> 
```

file: `a.txt` ; length=90 bytes
`a.txt:0`

serg3295, 2024-10-25:
I increased size of USB CDC RX buffer to 512 bytes and both the utilities and `upload-file.py` worked.

`CONFIG_ESP_CONSOLE_USB_CDC_RX_BUF_SIZE=512`

But I'm not sure that this solution was right.
Maybe I broke something 😐

serg3295, 2024-10-26:
#### esp32s2, CDC console.

A console utility (both nodemcu-tool and upload-file.py ) can only be performed once.
When trying to execute the utility a second time, there is no response from the esp.
An esp reboot is required. :cry: 

Steps to reproduce:

<details>
  <summary>file a.txt</summary>
    
```
1
1234567890
12345678901234567890
123456789012345678901234567890
12345678901234567890123456789012345678901234567890
123456789012345678901234567890123456789012345678901234567890
1234567890123456789012345678901234567890123456789012345678901234567890
12345678901234567890123456789012345678901234567890123456789012345678901234567890
123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890
1234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890

  ```
  
</details>

<details>
  <summary>upload-file.py</summary>
  
We are trying to upload a file using upload-file.py
  
  ```
❯ python ./upload-file.py -p /dev/ttyACM0 a.txt
Synchronising serial...
Uploading "a.txt" as "a.txt"
Sending loader...
Sending file contents...
Done.
❯ python ./upload-file.py -p /dev/ttyACM0 a.txt
Synchronising serial...
...
We're stuck here. CTRL+C
...
^CTraceback (most recent call last):
  File "/home/serg/Projects/python/nodemcu/./upload-file.py", line 170, in <module>
    if not sync(ser):
           ^^^^^^^^^
  File "/home/serg/Projects/python/nodemcu/./upload-file.py", line 124, in sync
    line = ser.readline()
           ^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 565, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt
  ```

  Perform hardware reset
  GOTO begin

</details>

jmattsson, 2024-10-26:
Can confirm weirdness on my S2 CDC-ACM. I have started looking into it, and I keep finding more weirdness 😩

serg3295, 2024-10-26:
esp32-s3; jtag console; make monitor

Is there a limit on the maximum length of a Lua string? 255 bytes + \n

```print('aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaasssssssssssssssssaaaaaafdfweerwefvsafwefasfrweaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaasssssssssssssssssssssssaaerereweradsfvaswdwefvaszzscfdwerawefwerwersdfasdfewdsdrewrewrewaaaaaaaaaaaaaaaaaaaasaaaaB')```

+1 byte causes error :disappointed:

jmattsson, 2024-10-27:
Regardless of the configured RX buffer size, the CDC-ACM is at risk of being overrun it seems. I've pushed an updated `upload-file.py` script which adds an ack scheme and chunks the file contents. It's tuned to work with the default 64 byte RX buffer, but can told to use a larger block size if so desired.

jmattsson, 2024-10-27:
And to answer your other question, yes, there is a max line length (`LUA_MAXINPUT`) which is set to 256. That could arguably be turned into a Kconfig option. Open another issue and tag me if that's something you'd like to see 😊

serg3295, 2024-10-27:
> It's tuned to work with the default 64 byte RX buffer, but can be told to use a larger block size if so desired.

I also use file upload/download in 256 byte chunks with acknowledgement. So there is no buffer overflow.
It might make sense to automatically increase the buffer size in sdkconfig to 512, if the CDC console is selected?

serg3295, 2024-10-27:
> That could arguably be turned into a Kconfig option. 

Lua string length of 256 bytes is quite enough.
It's just that the print(more than 255 bytes) command is executed on esp8266 Lua5.3
That's why I asked this question.

jmattsson, 2024-10-28:
@serg3295 It looks like the only way to change the default RX buffer is to _also_ automatically default to using the CDC-ACM instead of the UART. I'm not sure whether that's a sensible change in defaults?

serg3295, 2024-10-28:
You are right, there is no need to change the default value, since the USB CDC RX buffer size depends on the specific utility or IDE. In particular, for `nodemcu-tools` and `NodeMCU-Tool` the setting is `CONFIG_ESP_CONSOLE_USB_CDC_RX_BUF_SIZE=256`
I think the setting value should be specified in the documentation of an utility or IDE.

It might make sense to mention the setting `CONFIG_ESP_CONSOLE_USB_CDC_RX_BUF_SIZE` in `console.md`.

jmattsson, 2024-10-29:
Good idea, I've added extra notes to the module docs now. 👍

serg3295, 2024-10-31:
esp32s3; jtag console
An error occurs while writing LFS image.
I try to upload LFS image and write it using node.LFS.reload("lfs.img")

<details>
  <summary>crash log</summary>
  
  ```
rst:0x3 (RTC_SW_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375ab3
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1750
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d18
entry 0x403c9908
I (26) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (26) boot: compile time Oct 24 2024 11:38:11
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.1
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00180000
I (86) boot:  3 lfs              unknown          c2 01 00190000 00010000
I (93) boot:  4 storage          Unknown data     01 82 001a0000 00070000
I (101) boot: End of partition table
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=2bd24h (179492) map
I (146) esp_image: segment 1: paddr=0003bd4c vaddr=3fc98f00 size=042cch ( 17100) load
I (150) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=bb140h (766272) map
I (289) esp_image: segment 3: paddr=000fb168 vaddr=3fc9d1cc size=0032ch (   812) load
I (290) esp_image: segment 4: paddr=000fb49c vaddr=40374000 size=14e3ch ( 85564) load
I (314) esp_image: segment 5: paddr=001102e0 vaddr=600fe000 size=00064h (   100) load
I (323) boot: Loaded app from partition at offset 0x10000
I (323) boot: Disabling RNG early entropy source...
I (324) cpu_start: Multicore app
I (328) cpu_start: Pro cpu up.
I (331) cpu_start: Starting app cpu, entry point is 0x403756a0
I (0) cpu_start: App cpu up.
I (347) cpu_start: Pro cpu start user code
I (347) cpu_start: cpu freq: 160000000 Hz
I (347) cpu_start: Application information:
I (347) cpu_start: Project name:     nodemcu
I (347) cpu_start: App version:      tmr-libmain-binpatch150-891-ga0
I (348) cpu_start: Compile time:     Oct 24 2024 11:38:08
I (348) cpu_start: ELF file SHA256:  69144894737113bf...
I (348) cpu_start: ESP-IDF:          v5.1.3
I (348) cpu_start: Min chip rev:     v0.0
I (349) cpu_start: Max chip rev:     v0.99 
I (349) cpu_start: Chip rev:         v0.1
I (349) heap_init: Initializing. RAM available for dynamic allocation:
I (349) heap_init: At 3FCA1868 len 00047EA8 (287 KiB): DRAM
I (350) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (350) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (350) heap_init: At 600FE064 len 00001F84 (7 KiB): RTCRAM
I (351) spi_flash: detected chip: generic
I (351) spi_flash: flash io: dio
W (352) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (352) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (353) sleep: Configure to isolate all GPIO pins in sleep state
I (353) sleep: Enable automatic switching of GPIO sleep configuration
I (354) app_start: Starting scheduler on CPU0
I (354) app_start: Starting scheduler on CPU1
I (354) main_task: Started on CPU0
[... 141 lines cut ...]
I (454) wifi_init: rx ba win: 6
I (454) wifi_init: tcpip mbox: 32
I (454) wifi_init: udp mbox: 6
I (454) wifi_init: tcp mbox: 6
I (454) wifi_init: tcp tx win: 5760
I (454) wifi_init: tcp rx win: 5760
I (454) wifi_init: tcp mss: 1440
I (464) wifi_init: WiFi IRAM OP enabled
I (464) wifi_init: WiFi RX IRAM OP enabled

NodeMCU ESP32 build 2024-10-24 11:37 powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v5.1.3
cannot open init.lua: No such file or directory

  ```

</details>

esp32s2 breaks USB connection when trying to write LFS image.

However, in both cases the image is eventually written correctly and the program is executed from LFS.
```
> node.LFS.main()
Print from LFS
> 
```

jmattsson, 2024-11-01:
I am not (currently) able to reproduce the "once only" issue on my s2 here. I can upload multiple files just fine, using the default log settings (Info).

And yes, when the CDC console is active, WiFi initialisation interferes with USB enumeration somehow, as far as I can tell. Deferring wifi initialisation works around that issue, but results in the wifi log messages appearing after the Lua startup messages and prompt.

Can you reproduce the LFS issue on the main dev-esp32 branch too?

serg3295, 2024-11-01:
Unfortunately I still have the "once only" issue. I tested CDC console on esp32-s2 Mini and esp32-s3 Zero using the latest upload-file.py.

I'm on Linux Mint 22 x86_64 Linux Lenovo 6.8.0-48-generic

```
❯ python ./upload-file.py -p /dev/ttyACM0 a.txt
Loaded 521 bytes of file contents
Synchronising serial... ok
Uploading "a.txt" as "a.txt"
Sending loader........................................................................ ok
Waiting for go-ahead... ok
Sending file contents (using blocksize 80)....... ok, sent 7 blocks
Waiting for final ack... ok
Upload complete.
❯ python ./upload-file.py -p /dev/ttyACM0 a.txt
Loaded 521 bytes of file contents
Synchronising serial...!!!!!
NodeMCU not responding
```

I even tried running upload-file.py on Windows. Upload failed.

```
PS C:\Temp> python .\upload-file.py -p COM12 .\srrr.txt
Loaded 397 bytes of file contents
Synchronising serial... ok
Uploading ".\srrr.txt" as ".\srrr.txt"
Sending loader....................................................................Lua error:    stdin:65: invalid escape sequence near '".\s'
>   console.mode(console.NONINTERACTIVE)
>
```

I'll try to run upload-file.py on another computer later to rule out a hardware (software?) problem with the USB port.

What value is `CONFIG_ESP_CONSOLE_USB_CDC_RX_BUF_SIZE` set in your sdkconfig?

serg3295, 2024-11-01:
> Can you reproduce the LFS issue on the main dev-esp32 branch too?

We can not upload LFS image to ESP via usb-serial-jtag on the dev-esp32 branch, so there is no issue with node.reload() :smile: 

On esp32, esp32-c6 connected via `uart` console there are also multiple reboots with Guru Meditation error, but there is no freeze during reboot.
<details>
  <summary>crash log for esp32s2</summary>
  
  ```

mode:DIO, clock div:2
load:0x4086c410,len:0xd44
load:0x4086e610,len:0x2de4
load:0x40875728,len:0x17d0
entry 0x4086c410
I (26) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (27) boot: compile time Nov  1 2024 10:42:06
I (28) boot: chip revision: v0.0
I (30) boot.esp32c6: SPI Speed      : 80MHz
I (34) boot.esp32c6: SPI Mode       : DIO
I (39) boot.esp32c6: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00180000
I (83) boot:  3 lfs              unknown          c2 01 00190000 00010000
I (90) boot:  4 storage          Unknown data     01 82 001a0000 00070000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=420e0020 size=2f548h (193864) map
I (189) esp_image: segment 1: paddr=0003f570 vaddr=40800000 size=00aa8h (  2728) load
I (192) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=d0630h (853552) map
I (543) esp_image: segment 3: paddr=00110658 vaddr=40800aa8 size=199a0h (104864) load
I (592) esp_image: segment 4: paddr=0012a000 vaddr=50000000 size=00048h (    72) load
I (603) boot: Loaded app from partition at offset 0x10000
I (604) boot: Disabling RNG early entropy source...
I (615) cpu_start: Unicore app
I (616) cpu_start: Pro cpu up.
W (625) clk: esp_perip_clk_init() has not been implemented yet
I (632) cpu_start: Pro cpu start user code
I (632) cpu_start: cpu freq: 160000000 Hz
I (632) cpu_start: Application information:
I (635) cpu_start: Project name:     nodemcu
I (640) cpu_start: App version:      tmr-libmain-binpatch150-888-g7c
I (647) cpu_start: Compile time:     Nov  1 2024 10:42:00
I (653) cpu_start: ELF file SHA256:  4fb08325410851ec...
I (659) cpu_start: ESP-IDF:          v5.1.3
I (664) cpu_start: Min chip rev:     v0.0
I (668) cpu_start: Max chip rev:     v0.99 
I (673) cpu_start: Chip rev:         v0.0
I (678) heap_init: Initializing. RAM available for dynamic allocation:
I (685) heap_init: At 4081F9C0 len 0005CC50 (371 KiB): D/IRAM
I (691) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (698) heap_init: At 50000048 len 00003FA0 (15 KiB): RTCRAM
I (705) spi_flash: detected chip: generic
I (709) spi_flash: flash io: dio
W (713) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (726) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (737) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (745) sleep: Configure to isolate all GPIO pins in sleep state
I (752) sleep: Enable automatic switching of GPIO sleep configuration
I (759) coexist: coex firmware version: 77cd7f8
I (764) coexist: coexist rom version 5b8dcfa
I (769) app_start: Starting scheduler on CPU0
I (774) main_task: Started on CPU0
I (774) main_task: Calling app_main()

E (804) mmap: esp_mmu_paddr_to_vaddr(752): paddr isn't mapped
I (804) pp: pp rom version: 5b8dcfa
[... 150 lines cut ...]
I (824) wifi:config nano formating: disabled
I (834) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N
I (834) wifi:Init data frame dynamic rx buffer num: 32
I (854) wifi:Init static rx mgmt buffer num: 5
I (854) wifi:Init management short buffer num: 32
I (854) wifi:Init dynamic tx buffer num: 32
I (864) wifi:Init static tx FG buffer num: 2
I (864) wifi:Init static rx buffer size: 1700
I (864) wifi:Init static rx buffer num: 10
I (874) wifi:Init dynamic rx buffer num: 32
I (874) wifi_init: rx ba win: 6
I (874) wifi_init: tcpip mbox: 32
I (874) wifi_init: udp mbox: 6
I (884) wifi_init: tcp mbox: 6
I (884) wifi_init: tcp tx win: 5760
I (884) wifi_init: tcp rx win: 5760
I (894) wifi_init: tcp mss: 1440
I (894) wifi_init: WiFi IRAM OP enabled
I (904) wifi_init: WiFi RX IRAM OP enabled

NodeMCU ESP32 build 2024-11-01 10:41 powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v5.1.3
cannot open init.lua: No such file or directory
  ```

</details>

serg3295, 2024-11-01:
I uploaded _lfs.img_ to esp32c6; (esp_console:  usb_serial_jtag; dev-esp32 branch ), then `node.LFS.reload("lfs.img")`

<details>
  <summary>LFS reload log for esp32c6</summary>
  
  ```

NodeMCU ESP32 build 2024-11-01 11:12 powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v5.1.3
cannot open init.lua: No such file or directory
> node.LFS.reload("lfs.img")
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
--- 0x4001975a: software_reset_cpu in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd10
load:0x4086e610,len:0x2d08
load:0x40875728,len:0x17d0
entry 0x4086c410
I (22) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (22) boot: compile time Nov  1 2024 11:12:40
I (24) boot: chip revision: v0.0
I (25) boot.esp32c6: SPI Speed      : 80MHz
I (29) boot.esp32c6: SPI Mode       : DIO
I (34) boot.esp32c6: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  2 factory          factory app      00 00 00010000 00180000
I (78) boot:  3 lfs              unknown          c2 01 00190000 00010000
I (85) boot:  4 storage          Unknown data     01 82 001a0000 00070000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=420d0020 size=2f608h (194056) map
I (183) esp_image: segment 1: paddr=0003f630 vaddr=40800000 size=009e8h (  2536) load
I (186) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=cf980h (850304) map
I (530) esp_image: segment 3: paddr=0010f9a8 vaddr=408009e8 size=198d8h (104664) load
I (579) esp_image: segment 4: paddr=00129288 vaddr=50000000 size=00048h (    72) load
I (591) boot: Loaded app from partition at offset 0x10000
I (592) boot: Disabling RNG early entropy source...
I (593) cpu_start: Unicore app
I (595) cpu_start: Pro cpu up.
W (610) clk: esp_perip_clk_init() has not been implemented yet
I (611) cpu_start: Pro cpu start user code
I (611) cpu_start: cpu freq: 160000000 Hz
I (615) cpu_start: Application information:
I (620) cpu_start: Project name:     nodemcu
I (625) cpu_start: App version:      tmr-libmain-binpatch150-888-g7c
I (632) cpu_start: Compile time:     Nov  1 2024 11:12:35
I (638) cpu_start: ELF file SHA256:  ac361b6b82bfee71...
I (644) cpu_start: ESP-IDF:          v5.1.3
I (649) cpu_start: Min chip rev:     v0.0
I (654) cpu_start: Max chip rev:     v0.99 
I (658) cpu_start: Chip rev:         v0.0
I (663) heap_init: Initializing. RAM available for dynamic allocation:
I (670) heap_init: At 4081F840 len 0005CDD0 (371 KiB): D/IRAM
I (677) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (684) heap_init: At 50000048 len 00003FA0 (15 KiB): RTCRAM
I (691) spi_flash: detected chip: generic
I (695) spi_flash: flash io: dio
W (699) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (712) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (723) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (731) sleep: Configure to isolate all GPIO pins in sleep state
I (737) sleep: Enable automatic switching of GPIO sleep configuration
I (744) coexist: coex firmware version: 77cd7f8
[... 147 lines cut ...]
I (780) wifi:Init management short buffer num: 32
I (780) wifi:Init dynamic tx buffer num: 32
I (780) wifi:Init static tx FG buffer num: 2
I (780) wifi:Init static rx buffer size: 1700
I (780) wifi:Init static rx buffer num: 10
I (780) wifi:Init dynamic rx buffer num: 32
I (780) wifi_init: rx ba win: 6
I (780) wifi_init: tcpip mbox: 32
I (780) wifi_init: udp mbox: 6
I (780) wifi_init: tcp mbox: 6
I (780) wifi_init: tcp tx win: 5760
I (780) wifi_init: tcp rx win: 5760
I (780) wifi_init: tcp mss: 1440
I (780) wifi_init: WiFi IRAM OP enabled
I (780) wifi_init: WiFi RX IRAM OP enabled

NodeMCU ESP32 build 2024-11-01 11:12 powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v5.1.3
cannot open init.lua: No such file or directory
> 
> node.LFS.main()
Print from LFS
> 

```
</details>

serg3295, 2024-11-01:
> I am not (currently) able to reproduce the "once only" issue on my s2 here.

I connected the ESP32-S2 (using usb-cdc ) to a USB port on another machine and both `upload-file.py` and the utilities worked as expected, without errors.
I have very wierd USB ports on my Lenovo ideapad. :angry: 
Sorry for noice :confused:

jmattsson, 2024-11-18:
Are you getting the LFS reload errors even after you got the utilities to work? If so, maybe we should open a new ticket for that? Unless we can demonstrate that it's related to the transmission via the console, I mean.

serg3295, 2024-11-18:
Yes, LFS reload error is still present. I have opened a new issue #3667 .

jmattsson, 2024-12-10:
Closing this now that #3666 has been merged. Thank you for the patience!

PR #3666 (2024-12-10): Refactor to clean up and fix console handling
Fixes #3657.

- [x] This PR is for the `dev-esp32` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

In the interest of backwards compatibility I/we have in the past shoehorned the USB-Serial-JTAG and CDC-ACM console types into the `uart` module via the platform layer. That has kept adding complications and diffe [... cut]
closed 2024-12-10 by jmattsson