nodemcu/nodemcu-firmware #3660: wifi.connect() causes esp32c3 to reboot if log verbosity is set to "Info"

serg3295, 2024-08-09:
### Expected behavior

The ESP does not reboot after connecting to Wi-Fi if the "Log output" → "Default log detail" is set to "Info".

### Actual behavior

If `Component config → Log output → Default log verbosity ->` **"Info"** is set, the ESP reboots immediately after connecting to WiFi.

Everything works until this command.
[image]

If the default log verbosity level is set to **"Warning"**, the ESP connects to Wi-Fi and no reboot occurs. Because LOGI is not executed. 

Сurrent settings are:
(Top) → Component config → ESP System Settings → Channel for console output -> (X) Default: UART0
Channel for console secondary output (No secondary console) 

The error occurs with both UART and JTAG console. Also it occurs on IDF 5.2.2

<details>
  <summary>Crash log</summary>
    
```
I (415873) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
I (415913) wifi:mode : sta (f0:f5:bd:fc:36:e8)
I (415913) wifi:enable tsf
> I (416023) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (416023) wifi:state: init -> auth (b0)
I (416023) wifi:state: auth -> assoc (0)
I (416033) wifi:state: assoc -> run (10)
I (416043) wifi:connected with Dacha, aid = 12, channel 6, BW20, bssid = b0:4e:26:5c:eb:bd
I (416043) wifi:security: WPA2-PSK, phy: bgn, rssi: -40
I (416043) wifi:pm start, type: 1

I (416053) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (416053) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (416123) wifi:dp: 2, bi: 102400, li: 4, scale listen interval from 307200 us to 409600 us
I (416123) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (417073) esp_netif_handlers: sta ip: 192.168.1.74, mask: 255.255.255.0, gw: 192.168.1.1

assert failed: xQueueGenericSend queue.c:873 (!( ( pvItemToQueue == ((void *)0) ) && ( pxQueue->uxItemSize != ( UBaseType_t ) 0U ) ))
Core  0 register dump:
MEPC    : 0x40380706  RA      : 0x40380706  SP      : 0x3fcaad10  GP      : 0x3fc94800  
TP      : 0x3fc828a4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000042  S1      : 0x00000001  A0      : 0x00000000  A1      : 0x00000000  
A2      : 0x3fcaace8  A3      : 0x00000001  A4      : 0x40385d3a  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fcaaeb5  
S4      : 0x3fc967f8  S5      : 0x00000000  S6      : 0xffffffff  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fcaad10: 0x3c0d2a52 0x3c0d2c6c 0x3fcaad54 0x403889b8 0x00000001 0x000000f0 0x3c0d3038 0x40390232
3fcaad30: 0x3fc96bd0 0x3c0d3038 0x3fc96fbc 0x3c0d2a52 0x3fc96be0 0x3fcaad54 0x3fc96be4 0x3c0d2c6c
3fcaad50: 0x3fc967f8 0x00333738 0x3fca4ae0 0x65737361 0x66207472 0x656c6961 0x78203a64 0x75657551
3fcaad70: 0x6e654765 0x63697265 0x646e6553 0x65757120 0x632e6575 0x3337383a 0x28212820 0x70202820
3fcaad90: 0x65744976 0x516f546d 0x65756575 0x203d3d20 0x6f762828 0x2a206469 0x20293029 0x26262029
3fcaadb0: 0x70202820 0x65755178 0x3e2d6575 0x74497875 0x69536d65 0x2120657a 0x2028203d 0x73614255
3fcaadd0: 0x70795465 0x20745f65 0x55302029 0x29202920 0x3fcb0029 0x00065d31 0x3fcb0004 0x4038f194
3fcaadf0: 0x420081e4 0x000000f0 0x00000000 0x3fcaae1c 0xffffffff 0x00000000 0xffffffff 0x42091a52
3fcaae10: 0x3fcb0004 0x3fcaaf44 0x3fcb0004 0x00065d31 0x3c0e94f4 0x00000000 0x00000000 0x00000000
3fcaae30: 0x000000b7 0x0000004a 0xffffffff 0x00000000 0x00000000 0x00000000 0x3fcaa3e8 0x40388fec
3fcaae50: 0x000000b7 0x00000000 0xffffffff 0x00000000 0x00000000 0x3fcb0004 0x3fca3fa0 0x4038ddc0
3fcaae70: 0x00000000 0x3fcaa174 0x3fcb0004 0x00000000 0x00000000 0x3fcaaf44 0x3fcaa3e8 0x40389238
3fcaae90: 0x00000000 0x3fcaa174 0x00000000 0x420cd1e2 0x00000101 0x3c0e40e4 0x00000000 0x3fcb0004
3fcaaeb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaaed0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcaa174 0x420cd24c
3fcaaef0: 0x00000000 0x00000000 0x00000000 0x40389a46 0x00000000 0x00000000 0x00000000 0x00000000
[... 112 lines cut ...]
I (556) wifi:Init dynamic tx buffer num: 32
I (566) wifi:Init static tx FG buffer num: 2
I (566) wifi:Init static rx buffer size: 1600
I (566) wifi:Init static rx buffer num: 10
I (576) wifi:Init dynamic rx buffer num: 32
I (576) wifi_init: rx ba win: 6
I (576) wifi_init: tcpip mbox: 32
I (576) wifi_init: udp mbox: 6
I (586) wifi_init: tcp mbox: 6
I (586) wifi_init: tcp tx win: 5760
I (586) wifi_init: tcp rx win: 5760
I (596) wifi_init: tcp mss: 1440
I (596) wifi_init: WiFi IRAM OP enabled
I (596) wifi_init: WiFi RX IRAM OP enabled

NodeMCU ESP32 build 2024-08-09 13:18 powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v5.1.3
cannot open init.lua: No such file or directory
> 
```

</details>

### Hardware

esp32c3 supermini

serg3295, 2024-08-09:
I did some more research. This issue is common to all ESPs. ESP32 has the same error.
I built the firmware for ESP32 on commit "Add support for using multiple memory slots in rmt setup" with IDF v5.0.2 and the error is gone.

<details>
  <summary>Log for ESP32 on IDF 5.0.2</summary>
  
  
  ```
LFS image loaded
I (841) wifi:wifi driver task: 3ffc79dc, prio:23, stack:6656, core=0
I (841) system_api: Base MAC address is not set
I (841) system_api: read default base MAC address from EFUSE
I (861) wifi:wifi firmware version: 57982fe
I (861) wifi:wifi certification version: v7.0
I (861) wifi:config NVS flash: enabled
I (861) wifi:config nano formating: disabled
I (861) wifi:Init data frame dynamic rx buffer num: 32
I (871) wifi:Init management frame dynamic rx buffer num: 32
I (871) wifi:Init management short buffer num: 32
I (881) wifi:Init dynamic tx buffer num: 32
I (881) wifi:Init static rx buffer size: 1600
I (891) wifi:Init static rx buffer num: 10
I (891) wifi:Init dynamic rx buffer num: 32
I (891) wifi_init: rx ba win: 6
I (901) wifi_init: tcpip mbox: 32
I (901) wifi_init: udp mbox: 6
I (901) wifi_init: tcp mbox: 6
I (911) wifi_init: tcp tx win: 5760
I (911) wifi_init: tcp rx win: 5760
I (921) wifi_init: tcp mss: 1440
I (921) wifi_init: WiFi IRAM OP enabled
I (921) wifi_init: WiFi RX IRAM OP enabled
NodeMCU ESP32 build unspecified powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v5.0.2
cannot open init.lua: No such file or directory
> do  wifi.mode(wifi.STATION)
  wifi.start()
  wifi.sta.settxpower(8)
  wifi.sta.config({ ssid = "Dacha", pwd = "**", auto = true }, true)
  wifi.sta.on("got_ip", function(ev, info) print("NodeMCU IP config:", info.ip) end)
  wifi.sta.connect()
end>> >> >> >> >> 
I (8151) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07W (8151) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
I (8321) wifi:mode : sta (7c:9e:bd:07:6b:bc)
I (8321) wifi:enable tsf
> I (8371) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (9061) wifi:state: init -> auth (b0)
I (9071) wifi:state: auth -> assoc (0)
I (9071) wifi:state: assoc -> run (10)
I (9091) wifi:connected with Dacha, aid = 12, channel 6, BW20, bssid = b0:4e:26:5c:eb:bd
I (9091) wifi:security: WPA2-PSK, phy: bgn, rssi: -40
I (9111) wifi:pm start, type: 1
I (9121) wifi:AP's beacon interval = 102400 us, DTIM period = 2
NodeMCU IP config:	192.168.1.35
I (10121) esp_netif_handlers: sta ip: 192.168.1.35, mask: 255.255.255.0, gw: 192.168.1.1
I (14341) wifi:<ba-add>idx:0 (ifx:0, b0:4e:26:5c:eb:bd), tid:0, ssn:1, winSize:64

  ```
  
</details>

on IDF 5.1.2, 5.1.3, 5.2.2 we have the same bug

jmattsson, 2024-10-24:
I can confirm that I can reproduce this on an esp32c3 at least; I will investigate.

jmattsson, 2024-10-24:
@serg3295 This crash seems to be another stack overrun. Increasing the system event task stack size resolves it for me. Are you able to take it for a spin on your end too? The branch is [here](https://github.com/jmattsson/nodemcu-firmware/tree/dev-esp32-fix-wifi-log-info-crash) (I never remember how to fetch a PR 😅). Make sure you remove your old `sdkconfig` file to have the new default apply (or manually make the same change over in `Component config -> ESP System Settings -> Event loop task stack size`).

jmattsson, 2024-10-24:
Great, thank you for testing! I'll get that merged right away.

PR #3665 (2024-10-24): Increase default system event task stack size.
Fixes #3660.

- [x] This PR is for the `dev` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [n/a] The code changes are reflected in the documentation at `docs/*`.

The IDF defaults to a stack size of 2304. Bumping this to 3072 avoids the stack overrun crash during `ESP_LOGI()` in the `esp_netif_action_got_ip()` function during WiFi connect.
closed 2024-10-24 by jmattsson