nodemcu/nodemcu-firmware #3663: Increase LFS Capacity on ESP32 to Exceed 256kB Limit

SerafimPlugfield, 2024-10-09:
### Missing feature
With the current LFS (Lua Flash Store) implementation, there is a 256kB limit. From reading other issues, I understand that this limitation is a hardware constraint for the ESP8266 (please correct me if I'm mistaken). However, would this also be the case for the ESP32? Given that the ESP32 has become the more popular choice for many projects, it might be worth exploring whether a higher LFS capacity could be supported for it.

### Justification
The firmware we use is quite complex, and this limitation has made it increasingly difficult to add new features without significant refactoring or compromises. Increasing the LFS capacity could greatly ease firmware development and expand the possibilities for more advanced features in the future.

We would greatly appreciate any insight into this possibility. Thanks again for the amazing work and for supporting this fantastic project!

jmattsson, 2024-10-31:
I'm not across the internals of LFS, but could you see whether the fix on [this branch](https://github.com/DiUS/nodemcu-firmware/tree/refs/heads/dev-esp32-fix-5.3-lfs-maxsize) allows you to successfully build (and use) larger LFS images, please?

SerafimPlugfield, 2024-11-08:
I'm traveling right now, but I'll test it as soon as I can. Thanks