nodemcu/nodemcu-firmware #3667: LFS reload freezes on the PR #3666 firmware

serg3295, 2024-11-18:
### Expected behavior
LFS reloads without freezing

### Actual behavior
LFS reload freezes on PR #3666 firmware. 
However, the lfs image loads correctly and the program works after a hardware reset

### Test code

console channel: UART

```text
I (621) main_task: Calling app_main()

E (661) mmap: esp_mmu_paddr_to_vaddr(752): paddr isn't mapped

Erasing LFS from flash addr 0x190000.[0;31mE (681) task_wdt: esp_task_wdt_init(592): Invalid arguments[0m
 to 0x19ffff
LFS image loaded

the reload is stuck here!
```

console channel JTAG

```text
I (613) main_task: Calling app_main()

E (643) mmap: esp_mmu_paddr_to_vaddr(752): paddr isn't mapped

Erasing LFS from flash addr 0x190000.[0;31mE (663) task_wdt: esp_task_wdt_init(592): Invalid arguments[0m
 to 0x19ffff
LFS image loaded
E (723) usb_serial_jtag: usb_serial_jtag_driver_install(126): Driver already installed

the reload is stuck here!

```
### NodeMCU startup banner

<details>
  <summary>LFS reload log for esp32h2</summary>
  

====== build config:
lfs_size	65536
esp_console	usb_serial_jtag
ssl	true
modules	console,file,gpio,http,i2c,net,node,ow,pipe,tmr,uart
number_type	float

====== lfs:
lfs_size	65536
lfs_base	1638400
lfs_mapped	1108082688
lfs_used	2564

====== hw:
chip_minor_rev	1
flash_id	13123606
flash_size	4096
chip_model	esp32h2
cpu_cores	1
chip_major_rev	0
chip_features	80

====== sw version:
git_commit_dts	202410290121
node_version_minor	0
git_branch	dev-esp32-console
node_version	NodeMCU ESP32
node_version_revision	0
git_release	tmr-libmain-binpatch150 +895
git_commit_id	3579f52c1baea2e742f4cabbb8ce0d39d958f6e8
node_version_major	0

Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400031b6
SPIWP:0xee
mode:DIO, clock div:1
load:0x4083cfd0,len:0xd0c
load:0x4083efd0,len:0x2cfc
load:0x408460e8,len:0x1778
entry 0x4083cfd0
I (22) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (22) boot: compile time Nov 18 2024 12:35:25
I (24) boot: chip revision: v0.1
I (25) boot.esp32h2: SPI Speed      : 64MHz
I (29) boot.esp32h2: SPI Mode       : DIO
I (34) boot.esp32h2: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  2 factory          factory app      00 00 00010000 00180000
I (78) boot:  3 lfs              unknown          c2 01 00190000 00010000
I (85) boot:  4 storage          Unknown data     01 82 001a0000 00070000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=42090020 size=25620h (153120) map
I (174) esp_image: segment 1: paddr=00035648 vaddr=40800000 size=0a9d0h ( 43472) load
I (198) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=8804ch (557132) map
I (448) esp_image: segment 3: paddr=000c8074 vaddr=4080a9d0 size=0202ch (  8236) load
I (455) esp_image: segment 4: paddr=000ca0a8 vaddr=4080ca00 size=012c0h (  4800) load
I (459) esp_image: segment 5: paddr=000cb370 vaddr=50000000 size=00048h (    72) load
I (470) boot: Loaded app from partition at offset 0x10000
I (471) boot: Disabling RNG early entropy source...
I (475) cpu_start: Unicore app
I (478) cpu_start: Pro cpu up.
W (493) clk: esp_perip_clk_init() has not been implemented yet
I (494) cpu_start: Pro cpu start user code
I (494) cpu_start: cpu freq: 96000000 Hz
I (498) cpu_start: Application information:
I (503) cpu_start: Project name:     nodemcu
I (507) cpu_start: App version:      tmr-libmain-binpatch150-895-g35
I (515) cpu_start: Compile time:     Nov 18 2024 12:35:19
I (521) cpu_start: ELF file SHA256:  42ba0d13b08f6efe...
I (527) cpu_start: ESP-IDF:          v5.1.3
I (531) cpu_start: Min chip rev:     v0.0
I (536) cpu_start: Max chip rev:     v0.99 
I (541) cpu_start: Chip rev:         v0.1
I (546) heap_init: Initializing. RAM available for dynamic allocation:
I (553) heap_init: At 40810140 len 0003D240 (244 KiB): D/IRAM
I (559) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
I (567) spi_flash: detected chip: generic
I (571) spi_flash: flash io: dio
W (575) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (586) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (594) sleep: Configure to isolate all GPIO pins in sleep state
I (601) sleep: Enable automatic switching of GPIO sleep configuration
I (608) app_start: Starting scheduler on CPU0
I (613) main_task: Started on CPU0
I (613) main_task: Calling app_main()

E (643) mmap: esp_mmu_paddr_to_vaddr(752): paddr isn't mapped

Erasing LFS from flash addr 0x190000.[0;31mE (663) task_wdt: esp_task_wdt_init(592): Invalid arguments[0m
 to 0x19ffff
LFS image loaded
E (723) usb_serial_jtag: usb_serial_jtag_driver_install(126): Driver already installed

</details>

### Hardware
esp32h2

jmattsson, 2024-12-03:
Can reproduce on my C3 as well, I will attempt to get to the bottom of this.

serg3295, 2024-12-06:
I cherry-picked this PR to the `dev-esp32-console-module` branch and then tested on these hardware:

- esp32: uart
- esp32c3-mini: usb-serial-jtag
- esp32s2-mini: usb-cdc
- esp32s3-zero: usb_serial_jtag
- esp32s3-zero: usb-cdc
- esp32c6-nano: uart
- esp32c6-nano: usb_serial_jtag
- esp32h2-mini: uart
- esp32h2-mini: usb_serial_jtag

LFS image was reloaded successfully on all chips.

This PR also works on the `dev-esp32` branch. I tested on esp32 only.

jmattsson, 2024-12-06:
@serg3295 Thank you, much appreciated!

PR #3668 (2025-03-23): Issue restart rather than second lua_main() call on LFS reload.
By the time we get informed that there's an LFS reload we've already opened all the user libraries as well (and run their init code), so it's not safe to simply redo that with a fresh lua_State without also starting with a fresh hardware state. Currently attempting to do the second lua_main() call results in a hang, and a manual reset is required to progress. Doing the automatic restart is the correct approach here, even though this adds another reboot to the LFS reload process.

Also fixed up a couple of minor issues when embedde [... cut]
closed 2025-03-23 by jmattsson