nodemcu/nodemcu-firmware #3669: Wifi monitor doesnt start if called by anything on startup

freeman1216, 2024-12-08:
### Expected behavior
If called on startup should execute callback funtion accordingly
### Actual behavior
The file gets executed but the callback never called
If called using serial works fine
### Test code
init.lua
```Lua
uart.setup(0,9600,8,uart.PARITY_NONE,uart.STOPBITS_1,1)

tmr.create():alarm(1000,tmr.ALARM_SINGLE,function ()
  dofile("wifi.lua")
end)

```
```Lua
Wifi_table = {}
wifi.monitor.start(13, 0x80,function (pkt)
    if not Wifi_table[pkt.bssid_hex] then
        if  pkt[0] ~= nil then
          Wifi_table[pkt.bssid_hex] = pkt[0]
          uart.write(0,pkt.bssid_hex .. "," .. pkt[0].."\r\n")
        else
          Wifi_table[pkt.bssid_hex]="blank"
          uart.write(0,pkt.bssid_hex .. "," .. "blank".."\r\n")
        end
    end
end
)

local cleartimer = tmr.create()
cleartimer:alarm(4000000, tmr.ALARM_AUTO,function ()
  Wifi_table = {}
end)
```
### NodeMCU startup banner
NodeMCU 3.0.0.0 
        branch: release 
        commit: 36cbf9f017d356319a6369e299765eedff191154 
        release: 3.0.0-release_20211229 +25 
        release DTS: 202402250804 
        SSL: false 
 build 2024-12-07 03:17 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e) 
 build type: float 
        LFS: 0x0 bytes total capacity 
        modules: bit,file,gpio,node,ow,tmr,uart,wifi,wifi_monitor 

### Hardware
ESP8266 Nodemcu v2

freeman1216, 2024-12-08:
Found a workaround 
you need to set wifi mode to station first
init.lua
```Lua
uart.setup(0,9600,8,uart.PARITY_NONE,uart.STOPBITS_1,1)

wifi.setmode(wifi.STATION)

tmr.create():alarm(1000,tmr.ALARM_SINGLE,function ()
  dofile("wifi.lua")
end)

```

marcelstoer, 2024-12-09:
@pjsg What's your take on this? Is documenting that the WiFi mode should be set first good enough?