nodemcu/nodemcu-firmware #3677: esp8266 can't connect and get ip

johndoe71rus, 2025-02-09:
### Expected behavior
connect to wifi ap
### Actual behavior
in wifi ap log only
```
 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
 daemon.notice hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: did not acknowledge authentication response
 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
```
esp not receive ip from AP.

### Test code
simple wifi example code 
```Lua
wifi.setmode(wifi.STATION)
station_cfg={}
station_cfg.ssid="SSID"
station_cfg.pwd="pass"
station_cfg.save=false
wifi.sta.config(station_cfg)
wifi.sta.autoconnect(1)
wifi.sta.sethostname("NodeMCU") --Sets the NodeMCU hostname
print('ip: ',wifi.sta.getip()) -- prints the IP address of the NodeMCU 
```
print ip: null

### NodeMCU startup banner
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: release
	commit: 36cbf9f017d356319a6369e299765eedff191154
	release:
	release DTS: 202402250804
	SSL: false
	build type: integer
	LFS: 0x0 bytes total capacity
	modules: adc,bme280,bme280_math,encoder,file,gpio,i2c,mdns,net,node,ow,sjson,spi,tmr,tsl2561,uart,wifi
 build 2025-02-09 07:16 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
ESP8266 LiLyGO T-OI

###ADD
config openwrt AP
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'pci0000:00/0000:00:1d.7/usb2/2-1/2-1:1.0'
	option cell_density '0'
	option htmode 'HT20'
	option band '2g'
	option channel '6'
	option legacy_rates '1'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'SSID'
	option encryption 'psk2+ccmp'
	option key 'pass'
	option ifname 'wlan1'
	option ieee80211w '2'
	option wpa_disable_eapol_key_retries '1'
```

serg3295, 2025-02-09:
I think the issue will be closed due to #1010
I recommend asking the question on stackoverflow or (considering your nickname) on the amperka ru forum (in Russian) .

johndoe71rus, 2025-03-09:
fixed

 NodeMCU does not support option ieee80211w '2' ?

# ieee80211w: whether management frame protection is enabled
# 0 = disabled (default)
# 1 = optional
# 2 = required

set AP  option ieee80211w '1' and all work

closed 2025-03-09 by johndoe71rus