nodemcu/nodemcu-firmware #3678: net:close() fails - ESP32  ESP-IDF v5.3.2 Lua5.3

sabanora, 2025-03-16:
### Expected behavior

net:close() stops server

### Actual behavior

net:close() results in "Lua error: 	stdin:1: illegal argument"

### Test code

Wifi connected!
ftp_data = net.createServer(net.TCP, 30)
ftp_data:listen(21, function(s) print("listen") end)
ftp_data.close()

### NodeMCU startup banner
[0;32mI (31) boot: ESP-IDF v5.3.2 2nd stage bootloader[0m
[0;32mI (31) boot: compile time Jan 24 2025 18:21:39[0m
[0;32mI (31) boot: Multicore bootloader[0m
[0;32mI (35) boot: chip revision: v1.0[0m
[0;32mI (39) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (44) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (53) boot: Enabling RNG early entropy source...[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00180000[0m
[0;32mI (91) boot:  3 lfs              unknown          c2 01 00190000 00020000[0m
[0;32mI (99) boot:  4 storage          Unknown data     01 82 001b0000 00080000[0m
[0;32mI (106) boot: End of partition table[0m
[0;32mI (111) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=31f64h (204644) map[0m
[0;32mI (189) esp_image: segment 1: paddr=00041f8c vaddr=3ff80062 size=00008h (     8) load[0m
[0;32mI (190) esp_image: segment 2: paddr=00041f9c vaddr=3ffb0000 size=0440ch ( 17420) load[0m
[0;32mI (202) esp_image: segment 3: paddr=000463b0 vaddr=40080000 size=09c68h ( 40040) load[0m
[0;32mI (219) esp_image: segment 4: paddr=00050020 vaddr=400d0020 size=da148h (893256) map[0m
[0;32mI (526) esp_image: segment 5: paddr=0012a170 vaddr=40089c68 size=14fa4h ( 85924) load[0m
[0;32mI (559) esp_image: segment 6: paddr=0013f11c vaddr=400c0000 size=00064h (   100) load[0m
[0;32mI (575) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (575) boot: Disabling RNG early entropy source...[0m
[0;32mI (587) quad_psram: This chip is ESP32-D0WD[0m
[0;32mI (588) esp_psram: Found 8MB PSRAM device[0m
[0;32mI (588) esp_psram: Speed: 40MHz[0m
[0;32mI (591) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;33mW (599) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped[0m
[0;32mI (608) cpu_start: Multicore app[0m
[0;32mI (1490) esp_psram: SPI SRAM memory test OK[0m
[0;32mI (1498) cpu_start: Pro cpu start user code[0m
[0;32mI (1498) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (1498) app_init: Application information:[0m
[0;32mI (1501) app_init: Project name:     nodemcu[0m
[0;32mI (1506) app_init: App version:      1.4.0-master_20151229-906-gd53f[0m
[0;32mI (1513) app_init: Compile time:     Feb  1 2025 16:26:05[0m
[0;32mI (1519) app_init: ELF file SHA256:  ad64f9b89...[0m
[0;32mI (1525) app_init: ESP-IDF:          v5.3.2[0m
[0;32mI (1529) efuse_init: Min chip rev:     v0.0[0m
[0;32mI (1534) efuse_init: Max chip rev:     v3.99 [0m
[0;32mI (1539) efuse_init: Chip rev:         v1.0[0m
[0;32mI (1545) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1552) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1558) heap_init: At 3FFB9258 len 00026DA8 (155 KiB): DRAM[0m
[0;32mI (1564) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1570) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1577) heap_init: At 4009EC0C len 000013F4 (4 KiB): IRAM[0m
[0;32mI (1584) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator[0m
[0;32mI (1592) spi_flash: detected chip: generic[0m
[0;32mI (1595) spi_flash: flash io: dio[0m
[0;33mW (1600) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`[0m
[0;33mW (1610) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`[0m
[0;33mW (1621) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`[0m
[0;32mI (1630) main_task: Started on CPU0[0m
[0;32mI (1640) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (1640) main_task: Calling app_main()[0m

[0;31mE (1700) mmap: esp_mmu_paddr_to_vaddr(805): paddr isn't mapped[0m
I (1730) wifi:wifi driver task: 3ffd0028, prio:23, stack:6656, core=0
I (1740) wifi:wifi firmware version: b0fd6006b
I (1740) wifi:wifi certification version: v7.0
I (1740) wifi:config NVS flash: enabled
I (1740) wifi:config nano formating: disabled
I (1750) wifi:Init data frame dynamic rx buffer num: 32
I (1750) wifi:Init static rx mgmt buffer num: 5
I (1750) wifi:Init management short buffer num: 32
I (1770) wifi:Init static tx buffer num: 16
I (1770) wifi:Init tx cache buffer num: 32
I (1770) wifi:Init static rx buffer size: 1600
I (1770) wifi:Init static rx buffer num: 10
I (1780) wifi:Init dynamic rx buffer num: 32
[0;32mI (1780) wifi_init: rx ba win: 6[0m
[0;32mI (1780) wifi_init: accept mbox: 6[0m
[0;32mI (1790) wifi_init: tcpip mbox: 32[0m
[0;32mI (1790) wifi_init: udp mbox: 6[0m
[0;32mI (1790) wifi_init: tcp mbox: 6[0m
[0;32mI (1790) wifi_init: tcp tx win: 5760[0m
[0;32mI (1810) wifi_init: tcp rx win: 5760[0m
[0;32mI (1810) wifi_init: tcp mss: 1440[0m
[0;32mI (1820) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (1820) wifi_init: WiFi RX IRAM OP enabled[0m

NodeMCU ESP32 build 2025-02-01 16:04 powered by Lua 5.3.5 [5.3-int32-singlefp] on IDF v5.3.2

### Hardware
ESP32-WROVER incl. 8MB PSRAM

jmattsson, 2025-03-21:
Huh, LWIP's `netconn_close()` is returning arbitrary error codes now?!

jmattsson, 2025-03-21:
Okay, this is happening because the IDF/LWIP function `lwip_netconn_do_close` does not set a return code in this case, which means we just get random gibberish back as return code as that field is left uninitalised. Now the question is _why_, and why it's not actually closing the listening socket...

jmattsson, 2025-03-22:
@sabanora If you have the ability to, you may want to check my branch which should have a solution for this: https://github.com/jmattsson/nodemcu-firmware/tree/dev-esp32-netconn-close-fix

closed 2025-03-22 by marcelstoer

sabanora, 2025-03-22:
Works well!
Thx a lot!
Kind regards Rainer

PR #3679 (2025-03-22): Fix failure to close listening socket (#3678).
Fixes #3678 

- [x] This PR is for the `dev` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

The IDF LWIP's `netconn_close()` no longer returns valid error codes as the #ifdef'd code path the IDF uses does not end up actually doing much, which unfortunately includes actually setting the error code. As such, ch [... cut]